carla-simulator/carla #18: Vehicles missing textures

nsubiron, 2017-11-10:
Sometimes after adding the automotive materials and relinking them as in the documentation the textures fail to render.

Seems to be some weird issue with Unreal failing to recompile the shaders. Triggering a recompile solves the issue but doesn't seem to be an intuitive way of doing so (not forcing to recompile every shader which takes a long time).

Also seems to happen more often when you have several CARLA projects active in the same machine, probably because some intermediate files are shared among projects.

nsubiron, 2017-11-10:
A workaround could be to open "M_Carpaint" and move some nodes around then press "Apply", but yet it seems that some assets have to be opened in editor too to trigger the recompilation, (opening a car's skeletal mesh and the intermediate materials redirecting carpaint seems to work).

xevo-vstojkovic, 2017-12-07:
I have the same problem, but triggering the recompilation of shaders didn't resolve the issue. Using prebuilt binaries works fine. Any idea how to work around this?

nsubiron, 2017-12-11:
Sometimes the asset needs to be open in the editor when you recompile the shader. Try opening one or more of the cars' skeletal meshes at "Static/Vehicles/4Wheeled/", then trigger the recompilation again. It is a weird issue.

xevo-vstojkovic, 2017-12-11:
Thanks, I can confirm that deleting the `Unreal/CarlaUE4/DerivedDataCache` directory, running the editor and opening one of the skeletal meshes in `Static/Vehicles/4Wheeled` works.

nsubiron, 2018-04-03:
Closing this as we no longer use Epic's _Automotive Materials_.

closed 2018-04-03 by nsubiron