carla-simulator/carla #20: Download question??

s7ev3n, 2017-11-13:
I followed How to build CARLA on Linux, and stuck at Build Carla '$ ./Setup.sh', the code seems to download a Content_0.6.0.tar.gz from google drive, this file is huge, about 10GB. While I am in China, the connection is very unstable, and I downloaded almost a day, and failed. I saw the project page has 'Get the latest release here', I managed to download this, whose name is 'CARLA_0.6.0' , and its size after extraction is 6.3 GB. I am wondering what is CARLA_0.6.0? compare to Content_0.6.0.tar.gz? And is there any solution to download  Content_0.6.0.tar.gz in China? Thanks so much.

nsubiron, 2017-11-13:
CARLA_0.6.0 is the compiled project, which works on Ubuntu 16.04 or 17.04. So if you just want to try out CARLA, just run it.

Content_0.6.0 contains the raw assets needed to compile the project (3D models, textures, etc). They are only necessary if you want to compile CARLA from source and open the project in the Unreal Editor. The compiled project is smaller because the assets are compressed and optimized to be used from the binary version, not editable.

Given the bad connection in China, do you have suggestions on how to make download easier? One way would be to cut the file into smaller chunks, do you know of other ways?

dosovits, 2017-11-13:
@s7ev3n suggested he could upload to Baidu cloud storage, seems like a good idea, I'm now checking if there are issues with this.

nsubiron, 2017-11-13:
Keep in mind that we will upload a new content package with every new release

NikolausDemmel, 2017-11-13:
You could also try to extend `download_from_gdrive.py` to keep track of which chunks have been downloaded, such that it can resume an interrupted download. You will probably want to do some md5sum checking in the end.

dosovits, 2017-11-13:
@s7ev3n Ok, so we'll be happy if you redistribute CARLA in China, as long as it's done with proper attribution (mention that it's developed by CVC in collaboration with Intel and provide a link to carla.org). As @nsubiron pointed out, new releases will be coming out regularly, so you may need to re-upload from time to time to keep it up to date

@NikolausDemmel Thanks for the great suggestion! Do you know of some specific good way to do it in python? Some library you have experience with?

NikolausDemmel, 2017-11-13:
@dosovits, I don't really, sorry. After a quick google search, this seems simple enough: https://www.safaribooksonline.com/library/view/python-cookbook/0596001673/ch11s06.html

What is missing is maybe a md5check in the end to avoid corrupted downloads.

s7ev3n, 2017-11-14:
@dosovits Thanks, I will definitely mention your organization and carla.org, and I will try my best to keep my distribution up with new releases.

amiltonwong, 2017-11-14:
For quick solution, [here](http://pan.baidu.com/s/1qXASvpi) i what you want

s7ev3n, 2017-11-14:
@amiltonwong Hi, thanks so much, I have downloaded CARLA_0.6.0.tar.gz, this is the compiled project. my problem is running the ./Setup.sh, which tries to download a 10GB Content_0.6.0.tar.gz, have you downloaded this file? or maybe we cloud together help this redistribution?

amiltonwong, 2017-11-14:
No, just `tar zxvf CARLA_0.6.0.tar.gz` and then run it according to guidelines from [here](https://github.com/carla-simulator/carla/blob/master/Docs/how_to_run.md)

dosovits, 2017-11-14:
@NikolausDemmel , great, thanks!

@s7ev3n , awesome! Btw, if you send links to your Baidu Cloud files to carla.simulator email, we might be able to put links on the CARLA website.

@amiltonwong , thanks for the link!

amiltonwong, 2017-11-14:
@dosovits , you're welcome!
Ah, you will present CARLA in coming few hours @CoRL. I must watch it :)

s7ev3n, 2017-11-17:
@dosovits Sorry for the delay, I cut the Content_0.6.0.tar.gz into three pieces and now uploading it Baidu Cloud file, should be finished in two hours. Once it is done, I will email you the link. I am honored if you put the links on CARLA website. I succeeded compiling all the stuff, and writing my instructions about it in my article on Zhihu.com.

closed 2017-12-12 by dosovits

nsubiron, 2017-12-13:
I created #84 to have a single place to discuss possible solutions to the download problem.