carla-simulator/carla #21: Building on Windows

nsubiron, 2017-11-13:
Currently we do not have an official build system for Windows, however with some effort it is possible to get it working in this platform.

We will keep this issue with the progress we make for automating the installation of CARLA on Windows.

- - -

@yocabon provided some great PowerShell scripts: [Win64BuildScripts.zip](https://github.com/carla-simulator/carla/files/1467310/Win64BuildScripts.zip).

We are working for adding these scripts to CARLA. Please comment if you have any trouble using them.

- - -

#### Directions for building on Windows (not automated)

Install Unreal Engine 4.17 and Visual Studio 2015. (Later versions of Visual Studio work too).

Some Linux utilities are required (like make, cmake, rm…), it is best if you have some Linux environment like Cygwin installed.

The dependencies should be built and installed under "Util/Build" folder as follows

    Util/
      Build/
        boost-install/
        protobuf-install/

###### Build CarlaServer

Open a cmd.exe in the root folder of CARLA. Setup the environment calling vcvarsall.bat. If you have Visual Studio 2015 you can call it with (don’t forget the amd64 at the end)

    $ "C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\vcvarsall.bat" amd64

Then just run make and should work

    $ make

If everything goes well, CarlaServer should have been installed under "Unreal/CarlaUE4/Plugins/Carla/CarlaServer".

###### Build and launch CARLAUE4

At this point, double-clicking on "Unreal/CarlaUE4/CarlaUE4.uproject" should work. A warning about Carla plugin being not compatible appears sometimes, it’s an Unreal’s bug and can be ignored, press "no" to disable Carla plugin. Then press yes when asked to compile the missing modules.

If that doesn’t work, you can open the project with Visual Studio. Right click the "CarlaUE4.uproject" and select "Generate Visual Studio Project Files". Then open the Visual Studio solution it generates (*.sln). Hopefully this will give some hints on what fails.

jingleFun, 2017-11-16:
when Build CarlaServer with $ make
but How can I "make" in win10? I can`t find that command, and I get "makefile(8) : fatal error U1034" when I use nmake.

yocabon, 2017-11-16:
The powershell scripts provided should work without the make command. 
In order to use "make" you need a Linux like environment (cygwin...).

jingleFun, 2017-11-17:
sorry, I miss the script for build CarlaServer ^_^, Thank you so much ~

black3r, 2017-12-19:
suggestions for the powershell scripts:
- Install 7Zip with `-Scope CurrentUser`.. that way administrative rights are not needed.
- return to the original directory after successfully ending the script. Now when I run the ./Setup.ps1 I end up in /Util/Build and then I need to cd down to launch ./BuildServer.ps1

eds89, 2018-01-10:
Where can I download Unreal Engine version 4.17 exactly? I can only find 4.17.6 from the Epic Megagames installer.

black3r, 2018-01-10:
if the third version number is not specified, it should not matter, so 4.17.6 should be ok..

nsubiron, 2018-01-10:
@eds89 @black3r Yes, any subversion of 4.17 should work. If you want to go safe 4.17.2 has been tested.

eds89, 2018-01-10:
I'd be happy to go with the original 4.17, but I can't seem to find it on
the drop-down menu. Epic games seems to have removed it from some reason.

android-masaki, 2018-01-18:
It seems that I succeeded building with powershell, but BuildServer.ps1 couldn't work.
I got a message that is
"CMake Error: The source directory "C:/Users/nissan/Downloads/Win64BuildScripts/Util/cmake" does not appear to contain CMakeLists.txt."
Actually this file doesn't exist in this directory.
How can I deal with this>

nsubiron, 2018-01-19:
@android-masaki The PowerShell scripts are meant to be run inside CARLA project folder. First clone the CARLA project and copy the scripts in that folder.

josheligoldman, 2018-01-28:
Could somebody please just explain to me in simple words how to install Carla.

black3r, 2018-01-28:
1. Install the pre-requisites (Visual Studio 2015, Unreal Engine 4.17.*)
2. Download Carla
3. Download the PowerShell scripts from the top of this page
4. Make sure you've got ~40GB free HDD space
5. Extract Carla somewhere and extract the scripts into the same folder. There should be Setup.sh, Setup.ps1 and Setup.cmd next to each other. 
6. Run an elevated (admin) PowerShell prompt in the directory.
7. Run Setup.ps1 in the prompt
8. Run BuildServer.ps1 in the prompt
9. Download automotive materials from Unreal marketplace following the guide on wiki https://carla.readthedocs.io/en/latest/how_to_add_automotive_materials/
10. Open {CARLA}/Unreal/CarlaUE4/CarlaUE4.uproject in Unreal Engine. (You may encounter some errors or warnings, but you should be able to ignore them.). 
11. Press "Launch" (not "Play")
12. Wait until it builds (takes a loong time) and runs Carla
13. Now you've got Carla installed in `{CARLA}/Unreal/CarlaUE4/Saved/StagedBuilds/WindowsNoEditor` and can launch it with `CarlaUE4.exe` from that directory using the same console commands as described in the wiki for `./CarlaUE4.sh` 

If I've forgot something, maybe someone can correct me..., If anyone finds this guide to work, maybe we could put it on the wiki instead of "Coming soon"?

josheligoldman, 2018-01-29:
when I run Setup.ps1 I get the following error:
C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\cl.exe does not exist

nsubiron, 2018-01-29:
@black3r thanks!

@josheligoldman You need to [set up the environment to do a command-line build](https://msdn.microsoft.com/en-us/library/f2ccy3wt.aspx).

black3r, 2018-01-29:
@josheligoldman @nsubiron you don't need to setup the environment, the script does that for you.., and it looks like that environment setup part of the script is what doesn't work for you.., the thing is the script is hard-wired to use Visual Studio 2015, that has the `cl.exe` in folder `C:\Program Files (x86)\Microsoft Visual Studio 14.0\VC\bin\`

so if you want to use the automated script, you need to use Visual Studio 2015, and you have to have the C++ components installed (i think it is possible now to install only C#)

yocabon, 2018-01-29:
You either need Visual Studio 2015 or Visual Studio 2017 with visual C++ build tools installed.
By default, the scripts are looking for vcvars64.bat inside "C:/Program Files (x86)/Microsoft Visual Studio 14.0/VC". 

You change change that behavior by adding the argument -VCPath "YourDrive:/YourPath/Microsoft Visual Studio 14.0/VC".

Example
```
.\BuildServer.cmd -VCPath "D:/Programs/Microsoft Visual Studio 14.0/VC"

Building CarlaServer
Getting MSVC environment from D:\Programs\Microsoft Visual Studio 14.0\VC\bin\amd64\vcvars64.bat
...
```

razza-tu, 2018-01-29:
Hi everyone,

I have followed all of the above instructions in attempting this installation, but when trying to run the project, I am told that I am missing:

- UE4Editor-CarlaUE4.dll

- UE4Editor-Carla.dll

At which point of the build should these have been created? I have been using the PowerShell scripts provided at the top of this page.

davidSedlacek, 2018-01-30:
Hello everyone.

I had successfully build Carla on Win2012 server with Visual Studio 2015, thanks to the provided scripts 
( @nsubiron ) and then thanks to @black3r I prepared the Caral executable. -super :)

I would like to mention, that for the successful run of python client it is necessary to install several python libraries. I did not find them in any description. So I put the list here:

pip install pygame
pip install numpy
pip install protobuf
pip install pillow

nsubiron, 2018-01-30:
@razza-tu These two libraries are compiled by Unreal, when opening the project it usually asks if you want to rebuild them, just click "yes". Then the project will open after compilation if there were no errors.

@davidSedlacek True, this is mentioned in the [How to run](http://carla.readthedocs.io/en/latest/how_to_run/) guide, but not here. You could also get all dependencies at once with

    pip install -r PythonClient/requirements.txt

razza-tu, 2018-01-30:
@nsubiron Thanks for your reply. I have tried building these files with Unreal, exactly as you described. Unfortunately, my compiler crashes mid-build. The error reads as follows:

> C:\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\MapGen\DoublyConnectedEdgeList.cpp(47): fatal error C1001: An internal error has occurred in the compiler.
> (compiler file 'msc1.cpp', line 1507)
>  To work around this problem, try simplifying or changing the program near the locations listed above.

nsubiron, 2018-01-30:
@razza-tu Hmm yes, I've seen this before. Can you change in line 47 of "DoublyConnectedEdgeList.cpp" the `decltype(a)` to `float`? VS compiler doesn't like that.

```diff
--- a/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/DoublyConnectedEdgeList.cpp
+++ b/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/DoublyConnectedEdgeList.cpp
@@ -44,7 +44,7 @@ namespace MapGen {
     using Dcel = DoublyConnectedEdgeList;
     // from [-pi, pi] to [0, 1].
     auto normalize = [](auto a) {
-      constexpr decltype(a) twoPi = 2.0 * 3.14159265359;
+      constexpr float twoPi = 2.0 * 3.14159265359;
       a /= twoPi;
       while (a >= 1.0) a -= 1.0;
       while (a <  0.0) a += 1.0;
```
Let me know if it works and I will change the code.

razza-tu, 2018-01-30:
This fixed the problem for UE4Editor-CarlaUE4.dll.

However, UE4Editor-Carla.dll was still not created in this process. I am not given an error message for this. Just told that it couldn't be built.

When asked if I want to disable the plugin, if I respond "Yes", the project starts initializing. This process reaches 45% before telling me that UE4Editor-Carla.dll was not found, triggering an error.

nsubiron, 2018-01-30:
You are supposed to respond "No" to disable the plugin. That might be the issue.

If it keeps giving errors you can open the project with Visual Studio for debugging. Right click "Unreal/CarlaUE4/CarlaUE4.uproject"  and select _"Generate Visual Studio project files"_. Then open the generated .sln with Visual Studio and build the solution.

razza-tu, 2018-01-30:
I tried this initially. The log disappears before I can capture any information, and I am left with the message:

> CarlaUE4 could not be compiled. Try rebuilding from source manually.

josheligoldman, 2018-02-01:
Also, I am using visual studio community should this affect anything.

josheligoldman, 2018-02-01:
Sorry I thought I had installed the build
tools but I hadn’t no the problems fixed

josheligoldman, 2018-02-01:
Yay it’s building!!!!

eds89, 2018-02-01:
Hi,
I get the following error while running Setup.ps1

At D:\ *** \Setup.ps1:120 char:5
+     Install-Package 7Zip4Powershell
+     ~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (Install-Package:String) [],
CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

eds89, 2018-02-01:
Hi. The setup files apparently do not download boost, although Setup.ps1 returns "Build successful" at the end. Setup.ps1 downright fails.

josheligoldman, 2018-02-01:
when I launch the project it says it could not be compiled and to try building from source

josheligoldman, 2018-02-01:
@razza-tu have you solved it yet

eds89, 2018-02-01:
There are some problems with downloading the Content.tar.gz (a huge ~10GB
file) via the scripts. I had to download and unpack to Content/ manually.

eds89, 2018-02-01:
After buildserver.ps1 is over and I execute the .uproject file, I'm
prompted Carla had been built on a different engine version ans asks
whether I want to recompile it. I've clicked 'Yes' twice and, when I try to
launch Carla, Unreal Engine crashes at 95%.

Has anyone has the same problem?

Will try a third time and if the problem repeats, I'll compile it again and
do not re-compile it after.

josheligoldman, 2018-02-01:
@eds89 when I launch it I get the following error: 
CarlaUE4 could not be compiled. Try rebuilding from source manually

josheligoldman, 2018-02-02:
When I build with visual studio I get the following errors:
C:\Users\joshe\OneDrive\Documents\Github\carla\Unreal\CarlaUE4\Plugins\Carla\Source\Carla\Game\CarlaServer.cpp(18): fatal error C1083: Cannot open include file: 'carla/carla_server.h': No such file or directory
ERROR : UBT error : Failed to produce item: C:\Users\joshe\OneDrive\Documents\Github\carla\Unreal\CarlaUE4\Plugins\Carla\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-Carla.lib
C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.17\Engine\Build\BatchFiles\Build.bat" CarlaUE4Editor Win64 Development "C:\Users\joshe\OneDrive\Documents\Github\carla\Unreal\CarlaUE4\CarlaUE4.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.

josheligoldman, 2018-02-02:
When I run make in my Cygwin terminal, I get the following error:
CMake Error: The source directory "/cygdrive/c/Users/joshe/OneDrive/Documents/Github/carla/Util/cmake" does not exist.
which is weird because I the cmake folder does exist in that exact location.

josheligoldman, 2018-02-02:
Has anybody successfully built on windows 10

josheligoldman, 2018-02-03:
I have started fresh and when I finish running Setup.ps1, instead of getting 
Success,
I get,
****************
*** CARLA setup... - Done! ***
****************

eds89, 2018-02-13:
Has anybody had success generating a packaged version (Windows x64)? I don't and I get no error information. After the generation is apparently over, I get a totally empty directory. Has anybody experienced this same problem? I can launch CARLA in UEditor, but not out of it.

eds89, 2018-02-15:
Hi. While examining the generated log file, I've found some more errors. For instance:

# The asset filename being too long: This one happens multiple times.
[2018.02.13-17.08.03:297][419]UATHelper: Cooking (Windows): Cook: LogInit: Display: LogCook: Error: Couldn't save package, filename is too long :D:/auto_dev/CARLA/carla-0.7.1-src/Unreal/CarlaUE4/Saved/Cooked/WindowsNoEditor/CarlaUE4/Content/Static/Buildings/Bl_CityBuilding_REsidentialBuildingRoundCorners/N3/ResidentialBuildingRoundCorners_MainStructure_GroundfloorWindows_N2_Normal.uasset

# And. (This one happens only once)
Log file open, 02/13/18 16:29:11
LogWindows: Failed to load 'aqProf.dll' (GetLastError=126)
LogWindows: File 'aqProf.dll' does not exist

marcgpuig, 2018-02-15:
Hi @eds89,

For the first issue, we are currently changing certain file names because they are just too long.
A workaround could be using the project just in the root of your disk:

    D:/carla-0.7.1-src/Unreal/CarlaUE4

For the second error please try to remove your `Saved` and `Intermediate` folders, inside `Unreal/CarlaUE4`.

If this does not solve the problem, please check at this [issue](https://answers.unrealengine.com/questions/727614/unreal-engine-4181-crashes-on-launch.html?sort=oldest).

Hope it helps :)

eds89, 2018-02-15:
Right, I managed to package CARLA. However, the new Camera Effect I had
added does not seem to work. Although I have altered all *.cpp files that
mentioned "Settings" somewhere, I wonder whether there's any tutorial on
how to add new camera effects to Carla (as post-processing materials)?

Kind regards,

marcgpuig, 2018-02-16:
Hi @eds89,

Please take a look at these issues #125 and #130 and comment there if you have more doubts.
Specify what had you tried and we'll try to help you the best we can.

But please, let's keep this thread for **Windows specific** issues.

jcyriac2, 2018-02-25:
I'm trying to pass in the carla-settings.ini file to .\CarlaUE4.exe by executing
.\CarlaUE4.exe -carla-settings=path\to\ini\file.
I enabled networking by setting 'UseNetworking=true' but the server doesn't wait for a client to connect and starts up anyway. Am I passing in the arguments wrong?

marcgpuig, 2018-02-26:
Hi @jcyriac2,

We found a bug where command-line arguments fails to load absolute paths in both linux and windows.
Sorry for the inconvenience, we'll try to fix as soon as possible.

shriram17, 2018-04-02:
Hey, 
I'm using Unreal Engine 4.18 in Windows 10, I followed the steps mentioned and was after running the PS Script  Setup.ps1 in the prompt, I get 
_Compiling carla_server.proto...
CarlaServer/source/carla/server/: No such file or directory
Carla Setup... - Done_
Any ideas, why some of the files aren't being built correctly using the provided PowerShell scripts?

marcgpuig, 2018-04-04:
Hey @shriram17,

We will give more support for windows eventually but for now I suggest you to do it manually:
- Use teh Developer Command Prompt
- Clone the repo

      git clone

marcgpuig, 2018-04-04:
Hi @shriram17, sorry for the late response.

Unfortunately this process is not automated right now and we are working on it, meanwhile I recommend you to do it manually.

Necessary software:
[Git](https://git-scm.com/downloads), [Make](http://gnuwin32.sourceforge.net/downlinks/make-bin-zip.php), [Cmake](https://cmake.org/download/).
Be sure that these programs are added to your path, so you can use them from your command prompt.
Also [Unreal Engine](https://www.unrealengine.com/download) (v4.18.x) and [Visual Studio](https://www.visualstudio.com/downloads/) (2017 preferably).
- Clone the repository

      git clone https://github.com/carla-simulator/carla.git
- Download the assets from the version you need.
   + Build the download link with `https://drive.google.com/open?id=` followed by the choosen hash that you will find in [`Util/ContentVersions.txt`](https://github.com/carla-simulator/carla/blob/master/Util/ContentVersions.txt).
   + Unzip it in `Unreal/CarlaUE4/Content`. Create the `Content` folder if necessary.

- Dependencies:
+ Download and build **for win64**:
     * Boost 1.64
     * Protobuf 3.3.0
   
+ Put the generated binaries in:
     * `Util/Build/boost-install`
     * `Util/Build/protobuf-install`

     respectively
- In order to build CARLA you must **enable the x64 Visual C++ Toolset**. You have different options:
  + **Recomended:** Use [`Visual Studio x64 Native Tools Command Prompt`](https://docs.microsoft.com/en-us/dotnet/framework/tools/developer-command-prompt-for-vs).
  + [Enable a 64-Bit Visual C++ Toolset on the Command Line](https://msdn.microsoft.com/en-us/library/x4d2c09s.aspx) (the instructions will depend on the version of VS that you have).
- Once you have a console with the Visual Toolset, navigate into the `carla` folder where you cloned the repo and use:

      make

- If you have done everything alright, **carla server** must be compiled successfully in `Util/Build/carlaserver-build`.

- Double click `Unreal/CarlaUE4/CarlaUE4.uproject` and it will ask you to rebuild
     
      UE4Editor-CarlaUE4.dll
      UE4Editor-Carla.dll
  
  Agree.

- If there are problems generating the dlls, right click on `CarlaUE4.uproject` and select `Generate Visual Studio project files`, so you can try compiling from VS and check the errors.

- Success!

shriram17, 2018-04-05:
Hey @marcgpuig,
I followed the steps you mentioned above when using _make_ command in the VS x64 Native Tools Command Prompt (tried it with administrator privilege as well), I encounter this error 
> makefile:37: *** missing separator.  Stop.

shriram17, 2018-04-05:
Hey @marcgpuig 
I successfully compiled the **carla server**. On attempting to build CarlaUE4.uproject I encountered this error 
> error LNK2038: mismatch detected for 'RuntimeLibrary': value 'MT_StaticRelease' doesn't match value 'MD_DynamicRelease' in SharedPCH.Engine.h.obj,  (File : libprotobuf.lib)

The Protobuf 3.3.0 library was downloaded from the [official release page](https://github.com/google/protobuf/releases/tag/v3.3.0) and built following the steps mentioned in the [documentation](https://github.com/google/protobuf/blob/master/cmake/README.md). The files generated from _nmake install_ were placed in **protobuf-install** .
Are there any issues with the resources used or steps to generate the binaries? If so, I would appreciate it if you could share the right way to generate the binaries or resolve the issue.

marcgpuig, 2018-04-09:
Hi @shriram17, first of all, sorry, in linux we use protobuf v3.3.0, but in Windows better use **v3.3.2**.
While this is not your error it may fix some issues. I've updated the information in my previous comment.

I've made a [BAT script](https://gist.github.com/marcgpuig/57946f9b684f64e5f08487089c437ea3) that downloads and installs a ready to use protobuf build for CARLA.
Just put it in `Util/Build` and run it through a cmd with the **x64 Visual C++ Toolset enabled**.

Or if you want to do it yourself, add the following parameters at the cmake command while building protobuf:

```
-DCMAKE_BUILD_TYPE=Release
-Dprotobuf_BUILD_TESTS=OFF
-DCMAKE_CXX_FLAGS_RELEASE=/MD
-Dprotobuf_MSVC_STATIC_RUNTIME=OFF
```

Hope it help.

Liangsyl, 2018-04-22:
I had installed cygwin, and installed cmake, make,gcc ,gbd,git.
and i only have setup.cmd, and setup.ps1.
but  when i running setup.ps1, there is a problem showing like
`
Buding protobuf
Folder protobuf-source already exists, skipping git clone..
Getting MSVC environment from D:\Microsoft VS\INSTALL\VC\bin\amd64\vcvars64.bat
CMake Error at CMakeLists.txt:12 (project):
    Generator
        Unix Makefiles
    does not support platform specification, but platform
         x64
   was specified.

CMAKE Error: CMAKE_C_COMPILER not set, after EnableLanguage
CMAKE Error: CMAKE_CXX_COMPILER not set, after EnableLanguage
`
[image]

marcgpuig, 2018-04-23:
Hi @Liangsyl,

Please follow the instruccions on the previous comments and take a look at the documentation on [How to build CARLA on Windows](http://carla.readthedocs.io/en/documentation/how_to_build_on_windows/) (documentation branch).

You have installed cygwin that is some sort of _Linux Shell Emulator_, but then you are trying to run `.cmd` and `.ps1` files that are for Windows cmd and powershell that tries to generate a Microsoft Visual C++ (MSVC), while you have downloaded gcc, which is another compiler.

I strongly recommend you to follow the documentation that I provided to you and if you have some troubles just comment here and I will help you.

gtri-jjang38, 2018-04-23:
**If you are trying to install "Make" manually:**
go to [link](http://gnuwin32.sourceforge.net/packages/make.htm).

**or just download the installer from the Direct Link below (v3.81):**
["Make" installer for Windows](https://sourceforge.net/projects/gnuwin32/files/make/3.81/make-3.81.exe/download?use_mirror=astuteinternet&download=)

For Boost 1.64 and Protobuf 3.3.2 install, following is the structure that worked for me:

- Util/Build/boost-install
		-bin (contains b2, bjam)
		-boost (contains accumulators, algorithm, ... , weak_ptr)

- Util/Build/protobuf-install
		-bin (contains protoc.exe)
		-cmake
		-include
		-lib

Liangsyl, 2018-04-26:
Hi @marcgpuig ，
I followed the instructions from the documentation on How to build CARLA on Windows, and i downloaded git, make, and cmake, also added to system PATH. (GunWIN is the folder of MAKE)
[image]
and i have unreal enginen 4.18 and VS 2015(community)
then, i press the Win button and search for X64
[image]
the chinese means the native tools command prompt.
[image]
and then i downloaded Boost 1.64 and builded boost.
[image]
for building protobuf,  i download install_protobuf.bat and put it in Util/Build , and ran it. And 
[image]
I think i successfully built protobuf. thank for you showing me the structure @stephen-jang 
## 
**but when i tried to use MAKE,  there are a proble like** 
[image]
[image]
Could you help me to solve this problem? thank you so much.

LijieLiu, 2018-04-29:
Hey,
I have followed the documentation to build it on windows. Everything before launching CarlaUE4.uproject is perfect. But when I double-click CarlaUE4.uproject, it showed 
[image]
Then I clicked "yes", it showed 
[image]
So I deleted the four folders, Binaries and Intermediate in CarlaUE4 folder and carla folder, then I compiled carla server again, but I showed the same when launched CarlaUE4.uproject.
I right-clicked the uproject to generate visual studio project files and opened it by VS2017, then I build it but it showed 
[image] And when I press debug, it showed
[image]
Any method for that?
Thanks

marcgpuig, 2018-04-30:
Hi @LijieLiu,

please use Unreal 4.18 like we suggest in the documentation :)

marcgpuig, 2018-04-30:
Hi @Liangsyl,
Sorry for the late response, I've never faced an error like that, but looking into the last error you provide us, seems that you have `D:/CARLA/carla/Util/cmake` in your computer, but the generated project is searching in `D:/CARLA/carla/carla/Util/cmake`.

I think this is a good clue that you can investigate, because I'm not able to reproduce your error ;)

LijieLiu, 2018-05-01:
Hey @marcgpuig 
Thanks for reply
I have changed with version of UE to UE4.18.3, but still require to rebuild from source manually. Then I generated CarlaUE4.sln and build by VS2017, but cannot build.
[image]
Seems like some issue of boost and I think I have already build boost1.64.0 at boost-install file
[image]
So how can I solve that?
Thanks a lot

Liangsyl, 2018-05-02:
hi @marcgpuig , thanks for reply.
Now, I can open CarlaUE4.uproject in Unreal engine. And I am tring to connecting a python client, but i found out that i do not have CarlaUE4.exe in my carla folder which is cloned from https://github.com/carla-simulator/carla.git, and I tried to clone again, but the folder I cloned neither have CarlaUE4.exe in it.
I have no idea how can i connect to a python client, would you please help me?
Thanks a million!

[71 more replies]

closed 2020-03-26 by germanros1987