carla-simulator/carla #22: Where to find CarlaUE4.sh?

WenchaoDing, 2017-11-14:
I build carla from source successfully, but I cannot find CarlaUE4.sh in my project folder. I do not find this file in the released 0.6.0 version either. Is something wrong with my build?

nsubiron, 2017-11-14:
Hi,

"CarlaUE4.sh" is only present in the release version, inside the package [CARLA_0.6.0.tar.gz](https://github.com/carla-simulator/carla/releases/tag/0.6.0).

If you build CARLA from source, you must open CARLA project with Unreal Editor. This can be done with

    $ cd Unreal/CarlaUE4
    $ ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor "$PWD/CarlaUE4.uproject"

Adjust to match your Unreal Engine folder if necessary.

closed 2017-11-14 by nsubiron

krydm, 2018-05-02:
@nsubiron I've grabbed carla-0.8.2.tar.gz, and it doesn't contain CarlaUE4.sh either. On the other hand when I just open the project with ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor "$PWD/CarlaUE4.uproject", it opens fine, but the Pyhton client(/client_example.py --autopilot) returns me just
ERROR: (localhost:2000) failed to connect: [Errno 111] Connection refused.
I can't find the port setting in the editor(first thing I wanted to check) -- could you advise plz?
Besides it would still good to have the shell script -- is it possible?
I believe I follow the guidelines in http://carla.readthedocs.io/en/latest/
TIA

nsubiron, 2018-05-03:
@krydm The [CARLA_0.8.2.tar.gz](https://github.com/carla-simulator/carla/releases/tag/0.8.2) (compiled version) does have a "CarlaUE4.sh". If you grabbed the source code then it doesn't, in that case you need to compile dependencies and source code, open the Unreal project, enable networking in _"Unreal/CarlaUE4/Config/CarlaSettings.ini"_, and press _Play_ in the editor. But using the source code is only necessary if you want to modify and recompile CARLA, if you just want to run it you are just fine using the compiled version.

ishaan95, 2018-07-02:
Hi, 
I am trying to run carla on windows and I have built it using the automatic installation mentioned - 
[here](https://carla.readthedocs.io/en/latest/how_to_build_on_windows/)
I couldn't find the CarlaUE4.sh file in any folder. I followed the instructions in this thread and am not understanding the last instruction given by @nsubiron. I did compile the dependencies and source code but could not find CarlaSettings.ini at the location mentioned. 

Could you give more details as how to enable networking in the Carlasettings config file through the unreal editor, where the config file is located (Should we create our own config file like Example settings.ini?) and (just to clarify) what the sequence of commands to run carla in client server mode. 

Also, in client_example.py it is mentioned we can use make_carla_client context manager, where does that fit in all this?

marcgpuig, 2018-07-03:
Hi @ishaan95.

First of all, if you just want to run carla on Windows I recommend you to go to [release page](https://github.com/carla-simulator/carla/releases) and download the last Windows compiled version from there. Then you can run the .exe with the same parameters as CarlaUE4.sh (notice .sh files are for Linux), for example:

```
CarlaUE4.exe -carla-settings=Example.CarlaSettings.ini -windowed -resx=800 -resy=600
```

If you still want, not only use it, but modify the simulator with Unreal, please just copy `carla\Docs\Example.CarlaSettings.ini` to `carla\Unreal\CarlaUE4\Config\CarlaSettings.ini` and Unreal Engine will load the correct configuration from `CarlaSettings.ini` when you hit play on the editor.