carla-simulator/carla #23: Rebuild Error

zmonoid, 2017-11-15:
When running  ```UE4_ROOT=~/UnrealEngine_4.17 ./Rebuild.sh```

It will produce such error:
```
Using bundled libc++ standard C++ library.
Using fast way to relink  circularly dependent libraries (no FixDeps).
Performing 11 actions (8 in parallel)
[1/11] Compile PCH.CarlaUE4.h
[3/11] Compile Module.Carla.gen.cpp
[2/11] Compile Module.Carla.cpp
[4/11] Link libUE4Editor-Carla.so
clang: error: no such file or directory: '/home/bzhou/Github/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libc++abi.a'
clang: error: no such file or directory: '/home/bzhou/Github/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libboost_system.a'
clang: error: no such file or directory: '/home/bzhou/Github/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libprotobuf.a'
clang: error: no such file or directory: '/home/bzhou/Github/carla/Unreal/CarlaUE4/Plugins/Carla/CarlaServer/lib/libcarlaserver.a'
[5/11] Compile CarlaUE4GameModeBase.gen.cpp
[6/11] Compile CarlaUE4.init.gen.cpp
[7/11] Compile CarlaMapGenerator.gen.cpp
[8/11] Compile CarlaUE4GameModeBase.cpp
[9/11] Compile CarlaUE4.cpp
[10/11] Compile CarlaMapGenerator.cpp
```

I tried to find *.a files and softlink to corresponding folders but still produce this error

nsubiron, 2017-11-15:
It seems like the Setup.sh script failed to compile the dependencies. Can you run Setup.sh again and post the output?

closed 2017-11-17 by nsubiron

ecatixtsc, 2019-02-11:
When running make launch the following error occurs:

 -[install_boost]: [B2 ERROR] An error ocurred while installing using "b2.exe".
 -[install_boost]: Exiting with error...

Thanks

marcgpuig, 2019-02-11:
@ecatixtsc try the new branch `marcgpuig/windows_build`.

Rajshah05, 2020-02-12:
When running UE4_ROOT=~/UnrealEngine_4.17 ./Rebuild.sh
 
I am getting this error

bash: ./Rebuild.sh: No such file or directory