carla-simulator/carla #24: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.23'

onlytailei, 2017-11-15:
I'm usring ubuntu 14.04.
When I run carla (installed from the release version 0.6.0), I suffered the error as:
xxxxx/CARLA_0.6.0/CarlaUE4/Binaries/Linux/CarlaUE4: /lib/x86_64-linux-gnu/libm.so.6: version `GLIBC_2.23' not found (required by xxxxx/CARLA_0.6.0/CarlaUE4/Binaries/Linux/CarlaUE4)

I tried both in the system python3 and anaconda python3, both of them give the error.
Should I install the new glib?

nsubiron, 2017-11-15:
Hi, sorry, CARLA requires Ubuntu 16.04 or later.

closed 2017-11-15 by nsubiron

onlytailei, 2017-11-15:
Can it be used in docker?

nsubiron, 2017-11-15:
We haven't had time to test it yet, but take a look at issue #17

NikolausDemmel, 2017-11-15:
Just a note, #17 shows how to build everything in docker. Running UE4 and/or the editor in docker will probably be more involved to ensure access to graphics hardware and make the GUI work, but it might be possible...

micmarty, 2018-11-06:
Is there anyone who found solution for this problem.
I don't  root priviliges + manual glibc 2.23 build fails every time :1st_place_medal: 

Is there a possibility to build Carla with static link, not dynamic?

```bash
1)
$~/CarlaUE4/Binaries/Linux> ./CarlaUE4 
./CarlaUE4: /lib64/libm.so.6: version 'GLIBC_2.23' not found (required by ./CarlaUE4)

2)
$~/CarlaUE4/Binaries/Linux> ldd CarlaUE4 | grep libm.so.6
./CarlaUE4: /lib64/libm.so.6: version 'GLIBC_2.23' not found (required by ./CarlaUE4)
	libm.so.6 => /lib64/libm.so.6 (0x00007f296a90c000)
```

Thanks for any replies :D