carla-simulator/carla #25: A question related to A3C

JunhongXu, 2017-11-16:
Hi,

If I want to use A3C, do I need to open multiple instances and each instance is connected by a client? Or I can connect multiple clients to one instance?

dosovits, 2017-11-16:
Hi @JunhongXu ,

I ran multiple servers (a total of 12, 4 GPUs, 3 instances each), and connected a client to each. Connecting multiple clients to one server is interesting, and we plan to investigate this multiplayer setting at one point, but currently it's not supported.

Btw, we will soon release the code of all three methods we tested, after we clean it up a bit.

JunhongXu, 2017-11-17:
Thanks for your comment!

I have played around with the simulator a bit and got some questions. 

1. What is the maximum value for speed, collision, and acceleration? It is good to know about this because we need to normalize input/output when training a neural net.

2. Is it possible to capture a specific picture by passing location(x, y, z) and orientation(x, y, z) to the server? Or we need to drive the vehicle to that location and save the image? This is useful when we want to train our agent with different goal images at different episodes.

nsubiron, 2017-11-17:
> What is the maximum value for speed, collision, and acceleration?

They are not bounded, they are the result of the simulated physics and it depends on the vehicle you drive. You can play around with the keyboard passing `-carla-no-networking` to the server and see what are the usual values in your case. For instance, with the default Mustang it is very difficult to reach more 140km/h in our cities, as you will run out of city soon. But there is no hard-coded upper limit, it depends on the simulated capabilities of the vehicle. Same with collision, it's just a counter on how many collisions you have (weighted with collision intensity).

> Is it possible to capture a specific picture by passing location(x, y, z) and orientation(x, y, z) to the server? Or we need to drive the vehicle to that location and save the image?

We don't have a functionality to do that. At this point cameras are added to the vehicle only, you can add an arbitrary number of cameras at arbitrary poses but always relative to the player vehicle. You can also choose the start position of the player at the beginning of the episode, but from a list of predefined start positions. We want to extend CARLA to support arbitrary start positions in the future.

closed 2017-11-30 by nsubiron

nina124, 2017-12-26:
Hi, @dosovits 
Just want to know when the code will be released?
Look forward to that!

dosovits, 2017-12-26:
@nina124 , unfortunately things take longer than expected... There are already beta versions of trained imitation model (a separate repo in the carla-simulator organization) and benchmarking code (in a separate branch in the carla repo). Trained RL model will probably come some time in January. Training code may take quite a bit longer unfortunately. 

You may be interested to know that our colleagues from Nervana recently integrated CARLA with their RL framework https://github.com/NervanaSystems/coach , and even have some RL examples, I believe.

nina124, 2017-12-26:
@dosovits Thanks for your reply.
Definitely I am interested. I will look into this repo.

nina124, 2017-12-27:
Hi, @dosovits 
Another question, how to set which graphics card is used by OpenGL ?
Thanks!

dosovits, 2018-01-10:
@nina124 Sorry, was on vacation. If this is still of interest, it's described in "Running CARLA off-screen" in the docs http://carla.readthedocs.io/en/latest/how_to_run/ . It is a somewhat ugly approach, and we would be very happy if someone proposes a better one.

nina124, 2018-01-11:
@dosovits 
Thank you for your comment. However, I still don't know how to set specific graphic card, I followed the instruction in "[Running CARLA off-screen](http://carla.readthedocs.io/en/latest/how_to_run/)". It by default uses the GPU0. If I want to use GPU2, which step should I change in the six steps of "Running CARLA off-screen"?
I need to specify the GPU, because when training with A3C, multiple CARLAs will be running at the same time. Using the same GPU card to render them is impractical.