carla-simulator/carla #26: Question about the depth image format

WenchaoDing, 2017-11-16:
Hi,
I am writing a ros wrapper for carla client, almost finished... but I get confused about the format of depth image received from the client.

In your code, you do the post-processing like this for visualization purpose:
```
			self.depth_vec[0] = self.depth_vec[0][:,:,:3]
			self.depth_vec[0] = self.depth_vec[0][:,:,::-1]
```
and
```
	depth = depth.astype(np.float32)
	gray_depth  = ((depth[:,:,0] +  depth[:,:,1]*256.0 +  depth[:,:,2]*256.0*256.0)/(256.0*256.0*256.0 ))
```
we can see that you actually use the three channels of the received depth image. What is the physical meaning of the three channels? Why you normalize the three channels use different coefficients?

How can reconstruct the ground truth depth from your format (I mean for example 1.2m 3.3m etc for every pixel)?

WenchaoDing, 2017-11-16:
what about the intrinsic parameter of the capturing camera? so I can reproject the depth to 3d point cloud

marcgpuig, 2017-11-16:
Hi @WenchaoDing,  
This is due our depth codification, we use the 3 different channels to accomplish the maximum precision that we can get from a image format of that size.

### Depth Explanation
24 bit floating precision point codified in the 3 channels of the RGB.  
The order from less to more significant bytes is R -> G -> B.

R|G|B|int24||
-|-|-|-|-
00000000|00000000|00000000|0|min (near)
11111111|11111111|11111111|16777215|max (far)

Our max render distance (far) is 1km.

1. To decodify our depth first we get the int24.

        R + G*256 + B*256*256

2. Then normalize it in the range [0, 1].

        Ans / ( 256*256*256 - 1 )

3. And finally multiply for the units that we want to get. We have set the far plane at 1000 metres.

        Ans * far

### Depth visualization
In this case 
```python
self.depth_vec[0] = self.depth_vec[0][:,:,::-1]
```
we are getting the blue channel (with `[:,:,::-1]`) that is the most significant one, so you will see a computationally-inexpensive low-precision depth.

### Matlab implementation:
Here you have a matlab implementation in case any of you need to use it.

```matlab
function getDepth(ImagePath)
    % read the image
    im = imread(ImagePath);
    imshow(im);
    im = int32(im);
    
    % r -> Red channel
    % g -> Green channel
    % b -> Blue channel
    % f -> Far: the max render distance. 1000 meters in this case.

    % depth = ((r) + (g * 256) + (b * 256*256)) / ((256*256*256) - 1) * f

    depthMatrix = double( im(:,:,1) + (im(:,:,2) * 256) + (im(:,:,3) * 256 * 256) ) / double((256 * 256 * 256) - 1) * 1000;
    
    while true
        % mouse position
        [x, y] = ginput(1);
        x = floor(x);
        y = floor(y);
        % display depth
        disp([num2str(depthMatrix(y, x)), ' m']);
    end
end
```

**NOTE:** Thanks to your issue we noticed that we left this `- 1` on the client code, so the correct depth decodification is:

```python
gray_depth  = ((depth[:,:,0] + depth[:,:,1]*256.0 + depth[:,:,2]*256.0*256.0)/((256.0*256.0*256.0) - 1))
```
We are gonna fix this soon.

WenchaoDing, 2017-11-16:
Hi @marcgpuig ,

Thanks for your answer. Based on your answer, I get the true depth now. :)

BTW: we are actually reversing the channels (with [:,:,::-1]) so that blue is the MSB, instead of choosing the blue channel, aren't we?

marcgpuig, 2017-11-16:
Hey @WenchaoDing!  

Sorry, I misunderstood the code before. I edited the comment with the right answer.
The thing is that we get the image from Unreal Engine in **ABGR** format.

> what about the intrinsic parameter of the capturing camera? so I can reproject the depth to 3d point cloud

The camera parameters (FOV, position, and so on) are set by you in settings, please take a look at the example in the Carla documentation [here](http://carla.readthedocs.io/en/latest/carla_settings/). With these information you can build the intrinsic matrix.
We are working on the documentation of these topics. ;)

Thanks!

dosovits, 2017-11-16:
Hi @WenchaoDing , btw, would be great if you can share your ROS integration code when it's done, I'm sure many would be happy to use it ( see e.g. issue #12 )

WenchaoDing, 2017-11-17:
Hi @dosovits 

I wrote a ros wrapper for my quick development. So I didn't plan to write a very official, well-shaped one and submit a pull request for this. Anyway, you can find my simple [implementation](https://github.com/WenchaoDing/carla_ros.git) here if someone is interested.

dosovits, 2017-11-17:
@WenchaoDing , thanks! Well, hopefully one day it will grow and become better-shaped :)

s7ev3n, 2017-11-21:
@WenchaoDing Hi, thanks for your ros implementation. I have the same question about the camera intrinsic parameter, do you have any idea how to calculate it ? Thanks!

WenchaoDing, 2017-11-29:
Hi

@s7ev3n 

Based on FOV and image size you can compute it. Assume that the image is (800 * 600, FOV 100), you can simply get cx = 400, cy = 300, (assume fx = fy =f),   tan(FOV/2) = (1000/2) /f ==> f = 500/tan50 = 419.6...

WenchaoDing, 2017-11-29:
Hi

@marcgpuig ,

I found the z axis box extent for all the vehicles is strange. The z extent is always 0.32m, for all vehicles, no matter what type of the vehicle is. And the actually z size should be 0.32m * 2 = 0.64m? which is still too small.

nsubiron, 2017-11-29:
Hi @WenchaoDing,

Yes, the Z axis of vehicle's bounding boxes is always the same. The reason why is because we didn't care about the Z axis when we set up this boxes as they were meant to be used as 2D boxes for measuring the percentage of the car on other lane and off-road. Now that they are exposed to the client perhaps is worth to fit the height too, we will consider adding this.

marcgpuig, 2017-11-29:
Hi @WenchaoDing,
Regarding the intrinsic matrix, look at my comment https://github.com/carla-simulator/carla/issues/56#issuecomment-347957348

closed 2017-12-13 by nsubiron

TzabarDolev, 2023-09-19:
There is an small error in the formula which is given in the 'readthedocs' of the depth sensor (wrong order of color channels).
I wrote about it the CarlaSimBlog - https://carlasimblog.wordpress.com/2023/09/16/sensors/