carla-simulator/carla #27: carla_manual_control.py errors

nsubiron, 2017-11-16:
Running `carla_manual_control.py` gives the following errors
```
 Welcome to CARLA manual control 
 USE ARROWS for control 
 Press R for reset 
STARTING in a few seconds...
Successfully Connected to Carla Server
Staring Episode on Position  152
ERROR:root:'dict' object has no attribute 'iteritems'
Traceback (most recent call last):
  File "./carla_manual_control.py", line 220, in on_execute
    self.on_render()
  File "./carla_manual_control.py", line 196, in on_render
    self.labels_vec[0] = join_classes(self.labels_vec[0][:,:,2])
  File "./carla_manual_control.py", line 46, in join_classes
    for key,value in classes_join.iteritems():
AttributeError: 'dict' object has no attribute 'iteritems'
```

WenchaoDing, 2017-11-16:
It is due to python version. Refer to https://stackoverflow.com/questions/13998492/iteritems-in-python.

nsubiron, 2017-11-16:
Yes, all the client files have a `#!/usr/bin/env python3` at the top, this forces the execution to Python 3. An easy fix is changing this to Python 2, but it is probably better to fix the issues and keep Python 3 support.

dosovits, 2017-11-16:
@nsubiron Yes, let's definitely have both python2 and python3 support. I had this issue before, isn't it solved just by replacing iteritems with items?

commit 5ae7596d (2017-11-17): Fix #27, fix other issues with the manual control script too
closed 2017-11-17 by nsubiron