carla-simulator/carla #29: [build error] segmental fault Failed to produce item UnrealEngine_4.17/Engine/Binaries/Linux/libUE4Editor-SlateCore.so

YuehChuan, 2017-11-16:
*The main issue*:
[1/347] Link libUE4Editor-SlateCore.so
clang: error: unable to execute command: Segmentation fault (core dumped)
clang: error: linker command failed due to signal (use -v to see invocation)
ERROR: UBT ERROR: Failed to produce item: /home/user/UnrealEngine_4.17/Engine/Binaries/Linux/libUE4Editor-SlateCore.so
Total build time: 27.23 seconds (Local executor: 0.00 seconds)
Makefile:183: recipe for target 'UE4Editor' failed

Hi, thanks be so kind make this awesome simulator available.
I wonder someone fall into the same issue for building up Unreal engine.
Mine environment is clang  3.9.1; quite confirm  start from a clean ubuntu 16.04.2 .

Hardware details:  intel i7 with 12GB ram.  
Graphic card:
Asus ROG STRIX-GTX1080Ti-011G-Gaming
(Nvidia GTX1080ti)
mother board:GIGABYTE-GA-X58A-UD3R v2.(2010)
environment: ubuntu16.04.2

[image]

Thank you in advance.

YuehChuan, 2017-11-16:
With locate libUE4Editor-SlateCore.so
I found 
```bash=
/home/schwarm/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/libUE4Editor-SlateCore.so.response
```
under directory above
its not share object but a ASCII text
[image]
Am I doing somewhere wrong?

YuehChuan, 2017-11-17:
I found that this is describe in #17 by @NikolausDemmel. what I did add a new user name carla.
And I found this nice docs
https://github.com/carla-simulator/carla/blob/master/Docs/how_to_run.md
update a day ago. I will keep trying if I get the project run will update here.

YuehChuan, 2017-11-17:
[image]
new issue is while  `./carla_example.py --help` inside PythonClient/carla_manual_control.py 
found no python module 
```bash=
from .carla_server_pb2 import  SceneDescription,EpisodeStart,EpisodeReady,Control,Measurements,RequestNewEpisode
ImportError: No module named 'carla.protoc.carla_server_pb2'
```

YuehChuan, 2017-11-17:
All the nice docs here ! https://github.com/carla-simulator/carla/tree/master/Docs
Thanks!

nsubiron, 2017-11-17:
> ImportError: No module named 'carla.protoc.carla_server_pb2'

This is a common mistake, this file is generated while running the CARLA's Setup.sh script (it is generated by the Protobuf compiler). If you just want to regenerate this file afterwards you can run "./Util/Protoc.sh", but note that this requires the Setup.sh to have succeeded compiling protobuf.

> All the nice docs here ! https://github.com/carla-simulator/carla/tree/master/Docs

They are also available with nicer format at http://carla.readthedocs.io ;)

closed 2017-11-17 by nsubiron

YuehChuan, 2017-11-18:
@nsubiron  Now  stock in part ./Setup.sh  , It show cannot allocate swap memory. This issue is too specific maybe related to my hardware. I will check it out first. Anyway, thanks!  Awesome work ;P

YuehChuan, 2017-11-18:
after `UE4_ROOT=~/UnrealEngine_4.17 ./Rebuild.sh`
segmental fault
[image]

mslavescu, 2018-03-29:
I'm getting the same error as in the issue description, and I'm seeing these problems while running:

`~/carla$` UE4_ROOT=~/UnrealEngine_4.17 ./Rebuild.sh

> Writing project files... 92%
UnrealBuildTool Exception: System.ArgumentOutOfRangeException: startIndex cannot be larger than length of string.
Parameter name: startIndex
  at System.String.Substring (Int32 startIndex, Int32 length) <0x7f1427c5b780 + 0x000f4> in <filename unknown>:0 
  at System.String.Substring (Int32 startIndex) <0x7f1427c5b750 + 0x0001e> in <filename unknown>:0 
  at UnrealBuildTool.CodeLiteProject.WriteProjectFile (System.Collections.Generic.List`1 InPlatforms, System.Collections.Generic.List`1 InConfigurations) <0x403a8180 + 0x005fd> in <filename unknown>:0 
  at UnrealBuildTool.ProjectFileGenerator.WriteProjectFiles () <0x403a8000 + 0x000b0> in <filename unknown>:0 
  at UnrealBuildTool.ProjectFileGenerator.GenerateProjectFiles (System.String[] Arguments) <0x4158d5b0 + 0x01ff4> in <filename unknown>:0 
  at UnrealBuildTool.UnrealBuildTool.GuardedMain (System.String[] Arguments) <0x4153d000 + 0x02ffd> in <filename unknown>:0 
ESC[0;33mBuild CarlaUE4 project...ESC[0m
mono "/sharefolder/sdc/git/UnrealEngine_4.17/Engine/Binaries/DotNET/UnrealBuildTool.exe" CarlaUE4Editor Linux Development  -project="/home/test/carla/Unreal/CarlaUE4/CarlaUE4.uproject" 

And this at the end of :

> 4 errors generated.
[4/11] Compile CarlaUE4GameModeBase.gen.cpp
[5/11] Compile CarlaMapGenerator.cpp
[6/11] Compile CarlaUE4.cpp
[7/11] Compile CarlaUE4.init.gen.cpp
[8/11] Compile CarlaMapGenerator.gen.cpp
[9/11] Compile CarlaUE4GameModeBase.cpp
ERROR: UBT ERROR: Failed to produce item: /home/test/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
Total build time: 78.52 seconds (Local executor: 0.00 seconds)
Makefile:600: recipe for target 'CarlaUE4Editor' failed
make: *** [CarlaUE4Editor] Error 5

I use Docker and the instructions provided here by @NikolausDemmel:

[From source setup instructions: Additional dependencies](https://github.com/carla-simulator/carla/issues/17)

marintoro, 2018-04-30:
Hello,
I got the exact same problem than you @mslavescu. Did you find how to fix this error? (I am launching "UE4_ROOT=~/UnrealEngine_4.18 ./Rebuild.sh" cause I use UnrealEngine 4.18)