carla-simulator/carla #30: AI Client?

WenchaoDing, 2017-11-17:
Hi,

I am trying to reproduce the results in your paper, i.e., conditional imitation learning. I notice that you are collecting data mostly from your AI (I find the corresponding AI codes on the carla server). Is there any convenient way to modify the carla server code so that it loads the built-in AI, instead of listens to the client?  Thanks

nsubiron, 2017-11-17:
Hi,

Do you want to run the AI with or without the client?

If you want to collect data while the in-game AI drives, you should do it from the Python client. In the measurements package that the server sends every frame, it is included the `ai_control`. This `ai_control` is the vehicle control that the AI would do this frame. If you want to apply this control to the vehicle, you have to send it back to the server in the control message. We are working on the documentation of the [measurements](http://carla.readthedocs.io/en/latest/measurements) and making a cleaner API for Python, hopefully these things will be clearer soon.

If you mean running the server without a client, you should pass the command-line parameter `-carla-no-networking`. This way the vehicle is controlled from the keyboard (works like a standard videogame). Then you can enable the AI ("autopilot mode") by pressing "P". But in this mode you cannot extract any images or measurements, this mode is just for testing CARLA, drive around and take a look at the city.

WenchaoDing, 2017-11-17:
Hi @nsubiron 

I want to run with the client. :)

Thanks a lot for your answer. I didn't realize that there is AI control in measurements. Really amazing!

lucosanta, 2017-11-19:
Hi,

@WenchaoDing I have published a [gist](https://gist.github.com/lucosanta/ec88159922495f092bc15f51bbb62bf4) that use `ai_control ` for collecting data. It is really rudimental, but for me it works.

Feel free to try it and improve.

WenchaoDing, 2017-11-19:
Hi @lucosanta 

I wrote a similar one and got it work now. Thanks for your answer.

HI @nsubiron 

I am confused about the measurements of the traffic sign and speed limit.

In documentation, the description is

> Traffic light Contains its transform and state (green, yellow, red). Speed-limit sign Contains its transform and speed-limit.
 
When I print these states out, I get something like
`traffic_light position: (9498.998047,7041.000000,3821.324219)
sign:0.000000
traffic_light position: (10272.000000,5285.001953,3821.326172)
sign:2.000000
etc
`
It gives the whole list of traffic sign information on the map. But the position information is confusing (very large compared to vehicle position). And I cannot determine which traffic light or speed limit is taking effect.

felipecode, 2017-11-19:
@WenchaoDing 
The positions are expressed in centimeters ( As in Unreal ) that is maybe why it seems to be very large compared to the vehicle.
We have no yet released a way to determine which is the active traffic light. We leave this decision to the user. Yet, you can use the current car position  and orientation to have a good estimate on that.

Cheers.

closed 2017-11-28 by nsubiron

aakashsehgal, 2019-11-14:
Hi...any idea how could this be done in Carla 0.9.6 ?

commit d08ca65a (2017-11-24): How to add autopilot control, close #30, close #36