carla-simulator/carla #33: Get images from the game

Tak-Au, 2017-11-17:
I am able to get the server and client to run.  I loook though the python example.py.  I am trying to have the image and user action save so I can try out behavior cloning.  However I can’t seem to get image save although I set save image to true in the python example file.  I

nsubiron, 2017-11-17:
Yes, you are right, there is a bug in the current Python implementation. We will fix it asap. Sorry about this

Tak-Au, 2017-11-17:
Is there anything i can help to solve this problem?

closed 2017-11-18 by Tak-Au

nsubiron, 2017-11-18:
Hey, thanks for offering! At the moment we are discussing how to fix this, so not much you can help, but it's appreciated :) we'll come with a solution soon

Tak-Au, 2017-11-18:
I edited the carla_manual_control.py to allow for image and data recording.  I am currently after imitation learning.  My goal is for be able to interface with Keras model to allow for the model to drive the car.

naveen7v, 2018-03-26:
@nsubiron   I am unable to save the images too... what is the work-around to get the images ...

Thank you

nsubiron, 2018-03-26:
Hi @naveen7v,

You can check the [client_example.py](https://github.com/carla-simulator/carla/blob/master/PythonClient/client_example.py) to see how is done. You can save the images there by running with the `-i` switch

    $ ./client_example.py -i

Internally, the images received from the simulator have a member function to save the image as png

```python
measurements, sensor_data = client.read_data()

for name, measurement in sensor_data.items():
    measurement.save_to_disk('filename.png')
```

naveen7v, 2018-03-26:
@nsubiron 

Dear sir...the `client_example.py` is as you say and i ran it with the `-i` switch,
but the `client.read_data()` returns a empty `sensor_data` dict... but `measurements` are there

I just have rgb camera sensor enabled...is it a problem in my side?

Thank you

vnmr, 2021-03-23:
here