carla-simulator/carla #34: Installation error: Carla plugin not compatible with current engine version.

sandman, 2017-11-17:
Hi,
I'm following the instructions to get Carla up and running on an Ubuntu 16.04 machine. At the final step of manually linking the Automotive Materials package, I get an error when launching the CarlaUE4 project: 
[image]

Any advice how to proceed is much appreciated!

sandman, 2017-11-17:
When I proceed by selecting either yes/no, I get a further notification about some modules needed to be rebuilt for the correct engine version. Finally the compilation of CarlaUE4 fails. I've installed engine version 4.17 on Ubuntu. From windows, I downloaded the Automotive Materials for the 4.17 Engine Version.

nsubiron, 2017-11-17:
Hi,

Yes, this appears to be a bug appearing in Unreal 4.17, it happens to other plugins too. It is safe to ignore it, don't disable "Carla" plugin.

The second notification tells you that it needs to rebuild the code, in general you need to say "yes" to this one.

If compilation fails then something went wrong in the configuration process. Have you follow all the steps in the [docs](http://carla.readthedocs.io/en/latest/how_to_build_on_linux/)? Setup, then rebuild? Can you please do it again and get the log of the failing step?

  1. Run `./Setup.sh`, it should print "Success" at the end.
  2. Run `UE4_ROOT=~/UnrealEngine_4.17 ./Rebuild.sh`, it should launch the Unreal Editor at the end.

Afterwards you may need to relink the materials.

closed 2017-11-21 by nsubiron

sandman, 2017-11-27:
Hi,
Thanks for the tip and sorry for my delayed response.  Yes, I followed all the steps in the docs. After following your instructions for the rebuild, I still get an error when trying to build Carla. 

Step 1 is successful.
Step two fails with the below error: (it seems I need to rebuild the precompiled headers from UnrealEngine)

`[2/11] Compile Module.Carla.cpp
fatal error: file '/usr/include/linux/version.h' has been modified since the
      precompiled header
      '/home/sbg/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/sbg/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
fatal error: file '/usr/include/linux/version.h' has been modified since the
      precompiled header
      '/home/sbg/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/sbg/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'`

The  full logs are shared at https://hastebin.com/bucatehivu.md

Does it mean I need to uninstall UnrealEngine and reinstall it again from scratch? Appreciate any pointers,

nsubiron, 2017-11-27:
Hi,
Oh yes, this happens from time to time due to some system update. You need to recompile part of the Unreal code, it takes some time but that should fix it. You can force a recompile with the following

    cd Unreal/CarlaUE4/
    make CarlaUE4Editor ARGS=-clean
    make CarlaUE4Editor

sandman, 2017-11-27:
Hi,
Thanks a lot for your help...The build is successful! The Unreal Editor launch however gets stuck at 71% - from the logs it seems that the linker is unable to find some scenario files. You mentioned relinking the materials - how can I do this before the Unreal Editor launches? I downloaded and copied the AutomotiveMaterials package into the Content/ directory of the Carla folder (~/carla-master/Unreal/CarlaUE4/Content/) as mentioned in the docs.

Relevant logs at: https://hastebin.com/hoveheboga.md

Really appreciate your support here!

nsubiron, 2017-11-28:
Hi @sandman,
It is usual that the first time you launch Unreal Editor it takes quite a lot to start, it needs to recompile all the shaders. Check your CPU usage, there should be several ShaderCompiler instances, if that's so, just wait, eventually it will finish and launch the editor.

I can't see the logs at the link you gave.

sandman, 2017-11-28:
Hi @nsubiron ,
My CPU usage is still quite low (< 10% average acrosss 8 cores) and I don't see the ShaderCompiler instances in my process list.

I see several errors of the following type:
```
[2017.11.28-10.47.28:482][  0]Error: CDO Constructor (CarlaMapGenerator): Failed to find StaticMesh'/Game/Static/RoadLines/St_Road_XCross_LaneMarking.St_Road_XCross_LaneMarking'

[2017.11.28-10.47.28:482][  0]Error: CDO Constructor (CarlaMapGenerator): Failed to find StaticMesh'/Game/Static/RoadLines/St_Road_XCross_LaneMarking.St_Road_XCross_LaneMarking'
```
The logs are shared at https://pastebin.com/Gnj1jJKh

I will keep it running for long now and see how it goes.

nsubiron, 2017-11-28:
It seems like the assets didn't download. Can you check your Content folder? "Unreal/CarlaUE4/Content", there should be about 12GB of assets. If that's not the case, you can download them from [here](https://drive.google.com/open?id=1Gw8sw01hDEV4FtpHEZZwvS-6XN0jmaLT) and extract them in the folder.

sandman, 2017-11-28:
Funny, I just had ~ 1 GB of data from the Automotive Materials package downloaded from UE Launcher in Windows. I downloaded the assets from your link and now it works! Thanks a lot for your help! Now on to some exploring :)

nsubiron, 2017-11-28:
Great, have fun :)

vnmr, 2021-03-23:
zz

vnmr, 2021-03-23:
zz

vnmr, 2021-03-23:
we

vnmr, 2021-03-23:
qwe

vnmr, 2021-03-23:
three

vnmr, 2021-03-23:
three

vnmr, 2021-03-23:
three