carla-simulator/carla #36: how does carla autopilot actually work?

debdip, 2017-11-20:
As per carla paper description it's used 3 different approaches: Modular pipeline, Imitation learning, Reinforcement learning. In which approach applied in carla autopilot mode?

Is autopilot implementation is open source? I only found binary.

lucosanta, 2017-11-20:
I think that autopilot is an Unreal Engine pilot which moves through the streets. It is a test created just to move around the town environment. You can use it just requesting the measurements and picking `ai_control` data. You can look a similar example in this [gist](https://gist.github.com/lucosanta/ec88159922495f092bc15f51bbb62bf4).

nsubiron, 2017-11-20:
Hi @debdip, as @lucosanta says, by autopilot we refer to the hard-coded AI inside the game, this AI takes advantage of all the privileged information available in game and has nothing to do with the learning approaches mention in the paper and video.

debdip, 2017-11-20:
hello Mr. nsubiron and lucosanta. Thank you for your reply. 
That's why maybe semantic segmentation is disabled. 
I've another question Mr. nsubiron.
Is there any option available to make it enable so that I can perform PSPnet or others to apply and make decision?

nsubiron, 2017-11-20:
The AI control is sent every frame together with the measurements, you can send it back to the server to enable the autopilot, or modify if wanted to add for instance some noise to it. The gist that @lucosanta provided is a good example.

With the current API (0.6.0), this is how you get the AI control and send it back to the server
```py
measurements = carla.getMeasurements()
ai_control = measurements['PlayerMeasurements'].ai_control
carla.sendCommand(ai_control)
```

jiamings, 2017-11-21:
Does the API mean we can override `ai_control` and replace it with whatever we want?
Another question: can I have multiple clients that controls several cars in the same environment?

nsubiron, 2017-11-21:
> Does the API mean we can override ai_control and replace it with whatever we want?

Yes, you can modify it or replace with whatever you want. This is actually the way the learning methods in the paper control the vehicle.

> Can I have multiple clients that controls several cars in the same environment?

No, sorry, this is not yet supported. It is one of the features we would like to add in the future.

commit d08ca65a (2017-11-24): How to add autopilot control, close #30, close #36
closed 2017-11-28 by nsubiron