carla-simulator/carla #37: How to create your own maps?

UndeadBlow, 2017-11-20:
Hello. Please, help me to understand, how can I create new maps? There are Town 01 and 02 in RoadMaps by default, but if I want to try a track, for example? 
Is there some way to do that?

TheNihilisticRobot, 2017-11-20:
Hi! I don't know exactly what do you mean with trying a track. 
Our cities are created manually, the only "procedural" part is the first road layout, the grid.
Our road generation is a bit limited right now, very reticulated. We intend to give more organic options but this will take some time since we need this roads to maintain the extra information they give to the car.
If you could prescind from this extra information you could build any road you'd like using a few simple assets included with the proyect, the **SplineMeshRepeater** (Content/Blueprints/SplineMeshRepeater) with The model **TileroadTwolanes** (Content/Static/Road/TileroadTwolanes) or **ST_PathWay_Block** (Content/Static/Sidewalk) as the static mesh. We have a bit of Documentation in this topic: [Map customization](http://carla.readthedocs.io/en/dev/map_customization/). more In depth documentation about building your own environments is on the way.

UndeadBlow, 2017-11-20:
@TheNihilisticRobot 
Hi! Thanks for you work and your answer!
I meant with track something about motorway, like that.

For example, I found video (https://www.youtube.com/watch?v=fQ-2ds-aInU) which states that Carla was used. How did they create their map? Can Unreal Engine Level Editor helps me with that?

Also there are two .png files with map structure in projects, can't I edit that structure?

Also there is only one file in Content folder (CarlaUE4-LinuxNoEditor.pak) in pre-built package. Do I need to built from sources?

TheNihilisticRobot, 2017-11-20:
@UndeadBlow 
You are welcome!

So, this video is actually from another simulator from Udacity, not us, [whose car happens to be named Carla](https://medium.com/udacity/how-the-udacity-self-driving-car-works-575365270a40), like our simulator. 

The two .Png maps you see are generated from the existing in-game maps, and not the other way around. They store information about the lanes. There is no way to edit the maps from the binary build. You'll have to download the full project and build from source.

 You'll have every tool provided by Unreal Editor and the ones provided by us. To recreate something similar to the video you linked to us I suggest you use [Unreal's landscape tools](https://docs.unrealengine.com/latest/INT/Engine/Landscape/QuickStart/1/index.html) And our road 
 and vegetation assets (That I mentioned in my last message).

closed 2017-12-13 by nsubiron

owenustc, 2018-03-09:
whether the road of your map is follow opendrive xml format? Or there is a way that extract the road information from your map and convert to opendrive format?