carla-simulator/carla #39: Distance from Centre of Lane?

yanii, 2017-11-20:
Any pointers on where we would start if we wanted to collect something other than the current PlayerMeasurements? For example, I'd really like to be able to get the distance from the centre of the lane.

xl-sr, 2017-11-20:
Yes, I am interested in this too. 
Other measurements would be helpful as well, e.g. the output from the AI controller function IsThereAnObstacleAhead()

felipecode, 2017-11-21:
@yanii  That is not currently available. However on the next release we will give users access to a map. That will allow you to get the car position within the lane. This is probably helpful for your needs. To get access early watch the "new_client_api" branch. It will be there by today or tommorrow.

@xl-sr We try to provide only primary measurements. You could program the IsThereAnObstacleAhead function by getting the player's position and orientation and comparing with the position of all agents in the simulator.

nsubiron, 2017-11-21:
@xl-sr IsThereAnObstacleAhead() is an internal function fine-tuned for the current autopilot and its results may not be very intuitive. Like there are several magic numbers that just work with the current autopilot, only detects dynamic obstacles, and detects pedestrians way before they are on the road.

xl-sr, 2017-11-22:
ok thanks. I think the approach of comparing the agents positions will give me what I want to achieve.

WenchaoDing, 2017-12-05:
Hi @felipecode ,

I am writing some function to calculate the distance to the lane left boundary and the right boundary, and I notice your implementation:

` ori = self.map_image_lanes[int(pixel[1]), int(pixel[0]), 2]
  ori = ((float(ori) / 255.0)) * 2 * math.pi
  return (-math.cos(ori), -math.sin(ori)) `

I am confused about why you can obtain the lane orientation by reading from one channel of the image. Is there any format encoded? Could you please provide any hint on how to implement my function efficiently? Thanks in advance.

thepinguine, 2017-12-05:
@WenchaoDing
Indeed the orientation of the lanes is encoded in the third value (second index) of the RGB-value.
You can observe this, when you pick this value of a specific point of a turn in the road (e.g. "Town01Lanes.png") and perform a small variation.

nsubiron, 2017-12-05:
Yes, the image has encoded the information of the road
https://github.com/carla-simulator/carla/blob/60a114eaa291f69681b18546da37edb7c4000b52/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/MapGen/RoadMap.cpp#L50-L63

I was printing this for debugging purposes then @felipecode started using it in his planner :grinning:, just keep in mind that in the conversion to PNG may be some losses so the info won't be 100% accurate.

hs105, 2018-04-09:
I thinks "Distance from Centre of Lane" is available in the latest release (0.8)?

errorJY, 2018-05-09:
Hi,@WenChaoDing
How did you get the Distance from Centre of Lane.I want to get it but i don't know how to get it.

errorJY, 2018-05-29:
Hello,@nsubiron Nice to meet u again.
I have a problem that if I want to print the orientation of the lanes at screen（Fcolor[2]）.How should i do？

closed 2018-08-06 by stale[bot]

FitashUlHaq, 2021-05-18:
Hi, is there any update on this? I tried getting the distance between ego vehicle and way-point at the center of the road but no success

Dhagash4, 2021-06-13:
@FitashUlHaq You can get all the waypoints at the center using this so what I do is save the waypoints of the whole map in a text file and query from that text file the nearest waypoint from the car center

`world.get_map().get_waypoint_xodr().next_until_lane_end(1.0)`

This will give waypoint till the lane end 1m apart from each other

Hope this helps