carla-simulator/carla #42: Multible camera image  and disable traffic lights？

s7ev3n, 2017-11-21:
In the configuration file, I see we can set up multible cameras. I have a few questions:
1. What is the Intrinsic Camera Parameter of the cameras? I see we can set FOV.
2. If I set three cameras, how can I get image data through measuements? 
3. I have another question not related to cameras, can I disable the traffic lights?

felipecode, 2017-11-24:
Hello @s7ev3n . Thanks for your participation.
1. You can directly set FOV and resolution. Other parameters, such as lens distortion, are not currently
supported.
2. If you set three cameras like this:
```
[CARLA/SceneCapture/CameraLeft]
PostProcessing=SceneFinal
CameraPositionY=-15.0
CameraRotationYaw=-30.0
[CARLA/SceneCapture/CameraMiddle]
CameraPositionY=0.0
PostProcessing=SceneFinal
CameraRotationYaw=0
[CARLA/SceneCapture/CameraRight]
PostProcessing=SceneFinal
CameraPositionY=15.0
CameraRotationYaw=30.0
```
You will receive three BGRA cameras from the server that would be stored in your measurements dictionary. They always come in the same order as written on the file:
measurements['BGRA'][0] # Camera Left
measurements['BGRA'][1] # Camera Middle
measurements['BGRA'][2] # Camera Right
However, this interface will change a bit in version 0.7.0, we were putting some effort on having a more clear and fast interface. We will also provide documentation.

Regarding question 3:
Currently no. That would create chaos in the city ! But maybe is an useful and not so hard to add feature. We will consider for release 0.7.1 .

s7ev3n, 2017-11-25:
@felipecode Thank you so much !

closed 2017-12-13 by nsubiron