carla-simulator/carla #44: Measurements concerning traffic rules

lucosanta, 2017-11-21:
Hi,

In the paper related to the simulator, I have found this sentence:

> In addition to sensor and pseudo-sensor readings, CARLA provides a range of measurements associated with the state of the agent and compliance with traffic rules. Measurements of the agent’s state include vehicle location and orientation with respect to the world coordinate system (akin to GPS
and compass), speed, acceleration vector, and accumulated impact from collisions. Measurements
concerning traffic rules include the percentage of the vehicle’s footprint that impinges on wrong-way
lanes or sidewalks, as well as states of the traffic lights and the speed limit at the current location of
the vehicle. Finally, CARLA provides access to exact locations and bounding boxes of all dynamic
objects in the environment. These signals play an important role in training and evaluating driving
policies.

What are these measurements concerning traffic rules include the percentage of the vehicle’s footprint that impinges on wrong-way lanes or sidewalks, as well as states of the traffic lights and the speed limit at the current location of the vehicle? 
I want to train an A3C agent using CARLA and this feature would enable it. 

Luca

s7ev3n, 2017-11-21:
Hi,
You can check http://carla.readthedocs.io/en/latest/measurements/.

lucosanta, 2017-11-21:
Thank you very much!

closed 2017-11-22 by lucosanta