carla-simulator/carla #45: how to modify rewards function

sufengniu, 2017-11-21:
Hello, I wonder how to modify the rewards function, and where to check the defined rewards function.

Thanks

dosovits, 2017-11-21:
Hello @sufengniu , there are no pre-specified reward functions. When training an RL agent, the user is free to define the reward in any way based on the measurements the agent gets. In fact, defining a good reward function - balancing out things like safety, traffic rules and speed - is a very tricky problem by itself, and there is no single good answer yet.

In our paper http://proceedings.mlr.press/v78/dosovitskiy17a/dosovitskiy17a.pdf , when evaluating the methods we count 1) the percentage of successfully reached goals and 2) Average distance driven between two traffic rules violations. We will release our evaluation code within a few weeks.

sufengniu, 2017-11-22:
Thank you! will close this issue, I may re-open the issue if I have some difficulties for the self-define rewards

closed 2017-11-22 by sufengniu

kbarbora, 2019-07-25:
Hi Sufengniu, did you were able to create a self-define rewards system in carla?