carla-simulator/carla #46: Creating training data and grabbing steering angles

farzaa, 2017-11-21:
Hi all!

If I wanted to create my own driving dataset by driving in Carla manually myself (ex. for an hour), would I be able to grab the exact steering angle? Right now, [this](https://github.com/carla-simulator/carla/blob/master/PythonClient/carla_manual_control.py) script obviously would let me do this. But, it seems the steering angle is set to either -1 or 1 depending on if I'm holding down A or D. 

`        if keys[K_LEFT] or keys[K_a]:
            control.steer = -1.0
            pressed_keys.append('left')
        if keys[K_RIGHT] or keys[K_d]:
            control.steer = 1.0
`

So, this would mean I would lose all those angles in between, correct? So my ground truth for steering angles would be just a bunch of -1's and 1's. Is this intended?

felipecode, 2017-11-22:
Hey farzaa.
That is correct ! 
Well, this happens because of limitations of using a keyboard ( That keys are either pressed or not pressed)
If you have a different joystick, with sensitive buttons, you could access a continuous value and also use pygame.
Also consider using the in game AI for data collection , please check #30 .

Tak-Au, 2017-11-23:
I modify the manual control file to take mouse input so you can steer left or right with the mouse.  This will give you continuous steering angle value.

closed 2017-11-30 by nsubiron

adiveloper, 2018-10-22:
How do I record data any idea ?

adiveloper, 2018-10-22:
I am new to carla. can someone please explain me how to start recording data(camera and steer,brake) train the data and then run my model based on training?
A very very brief overview will be highly appreciated....

Fangwq, 2019-01-29:
I'm new to Carla. I want to gather the data during each driving simulation and use it to train my own model. How to do it ?

haoliHuan9, 2022-07-21:
> I am new to carla. can someone please explain me how to start recording data(camera and steer,brake) train the data and then run my model based on training? A very very brief overview will be highly appreciated....

hi i am just wondering if you solve this problem to record your own data like steer angle, brake.. and i will be very appreciate if you can share your solution. Thanks a lot!