carla-simulator/carla #47: Question, How to train a car to move from one point to another point in town?

Tak-Au, 2017-11-21:
I want to train a car that can move through town from one location to another.  Is there a way to get a map of a town?   How does one actually approach this problem?   

Tak Au,

lucosanta, 2017-11-21:
Hi,

They have implemented it in [new_client_api branch](https://github.com/carla-simulator/carla/tree/new_client_api). Look at [carla_manual_control.py](https://github.com/carla-simulator/carla/blob/new_client_api/PythonClient/carla_manual_control.py). you will see a `from carla.planner.map import CarlaMap` where there is an example of usage.

lburzawa, 2017-11-28:
Is the stuff in branch "clientdevbranch" working? I am trying to run "carla_planner_example.py" but it gives me error "ImportError: No module named 'carla.protoc.carla_server_pb2' ".

nsubiron, 2017-11-28:
No, the new python API is already in dev branch, we expect to release it today to master branch together with a new CARLA binary :)

felipecode, 2017-11-28:
@lburzawa , the clientdevbranch is related to the features to be added in a few weeks.
Not currently functional.

closed 2017-11-28 by nsubiron