carla-simulator/carla #50: Any plans for more features like additional sensors, vehicle model, etc.?

sibojia, 2017-11-24:
It's really a great work of autonomous driving simulator. I'm currently working on a simulation platform for autonomous driving algorithm testing, thus a lot of new features are needed for this purpose. Some of them includes: additional sensors e.g. lidar, radar and multiple cameras, map generation by script, control of other cars, vehicle dynamic modelling, etc. 
I want to know if there is a detailed plan for the future development of CARLA, what will be added and what won't, and when it will be ready. I believe this helps for me and other users to contribute codes and make it better together. Thanks very much!

felipecode, 2017-11-24:
Hello @sibojia. Thanks a lot for your interest.
We are now going through intense meetings for building the CARLA roadmap.  There will be a detailed roadmap published soon on the website and
github. We are also studying how to get the community more involved.

What I can advance to you is that, on the things you cited:

- Multiple cameras: we already support.
- Lidar and flow: To be added definitely. 
- Map generation script: Already on the way, there will be ways for community to add their maps, but completely autonomous generation is unfeasible in near future.
- Control of other car: One of our main objectives, for sure to be added.
- Vehicle dynamic modelling: A bit out of our roadmap, but an interesting feature. We could study on eventually geting help of the community for that.

Thanks again for your interest.

sibojia, 2017-11-25:
Thanks for your reply, I'm looking forward to the future plan!

nsubiron, 2017-11-28:
We added the roadmap to the readme and create a [board](https://github.com/carla-simulator/carla/projects/2) for keeping track of it.

closed 2017-11-28 by nsubiron

jobino, 2018-07-16:
Hello guys, I agree with @sibojia and thank the community for CARLA, how could I develop the RADAR sensor or any other sensor?

marcgpuig, 2018-07-16:
Hi @jobino.
Sadly , we do not have any tutorial for this...
Take a look into the lidar implementation for an example :)

gruberto, 2018-08-06:
Hi,
I want to plug an additional sensor model for a gated camera that requires albedo and depth information.  If I understand it correct, this would be an additional post-processing and must be changed in the C++ Code under Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Sensor. Can you give me some hint where and how the C++ code and the python api are linked? Thank you.

Regards,
Tobias

marcgpuig, 2018-08-20:
Hi @gruberto.

First, you must make your own shader (Unreal post-processing material) to achieve the final image you are searching for. Thankfully for you, UE4 provides a really good node editor to create your own materials easily. Take a look at our materials as an example (see #317).

Once you have your post-processing material working, it should be the same implementation as the other sensors.

Cheers!

nievsamolina, 2018-10-04:
Hello! 
Great work with the simulator. I would like to know if it is possible to create my own vehicle model (with some specific vehicle dynamics) and use it as a player in CARLA; if it is possible, any clue of how to do it? or maybe can I connect a Vehicle model made in Simulink as an input for the Sim?

If this feature is not avilable yet, is it possible to have the parameters of the vehicle dynamics of the 16 vehicles that are already implemented in the simulator?

Thank you very much!
Niev

sh-shu, 2018-11-13:
@nievsamolina 
I am also interested. Since matlab/simulink is the common platform for algorithms, does Carla develop the interface for it? Or is there any user-defined branch which has this kind of functions?