carla-simulator/carla #53: Problem when building on linux

Jacquesval, 2017-11-27:
When I follow the instructions: How to build on linux, I get the following error:
Failed to build Boost.Build build engine
Consult 'bootstrap.log' for more details
And when I look in bootstrap.log, I see the following line:
./build.sh: 17: ./build.sh: clang: not found
but clang is already installed on my machine, so I don't understand the problem.
Can you help me please ?

Jacques

NikolausDemmel, 2017-11-27:
What happens when you run

    clang --version

in the terminal?

Jacquesval, 2017-11-27:
Ok my bad I thought it was installed because we need it to build the unreal engine, but apparently not. Now it says that the version is clang 3.8 even if I installed clang 3.9 to build the unreal engine.

nsubiron, 2017-11-27:
You may need to set clang 3.9 as default clang version with

    sudo update-alternatives --install /usr/bin/clang++ clang++ /usr/lib/llvm-3.9/bin/clang++ 100
    sudo update-alternatives --install /usr/bin/clang clang /usr/lib/llvm-3.9/bin/clang 100

Jacquesval, 2017-11-27:
Ok thanks, it seems to work now.

closed 2017-11-27 by dosovits