carla-simulator/carla #56: How to get camera intrinsic matrix?

syinari0123, 2017-11-28:
Thank you for publishing this great drive simulator code.
I am trying to map depth points(u,v,depth) from Pixel coordinates to World coordinates(x,y,z).
To do this, Camera intrinsic parameter is necessary. I found that "CameraFOV(horizontal field of view) and ImageSize (in ```CarlaSettings.ini```)" is parameters for calculating camera intrinsic matrix. I thought as following.
```
Focus_length = ImageSizeX /(2 * tan(CameraFOV * π / 360))
Center_X = ImageSizeX / 2
Center_Y = ImageSizeY / 2
```
But I found this intrinsic matrix is not correct enough because 3D points (transformed from depth points(u,v,depth) by this matrix) has some distortion. 

Could you tell me how to get camera intrinsic matrix? 

(Now I think Focus_length_per_pixel (fx, fy) or distortion parameter is necessary for this. )

marcgpuig, 2017-11-29:
Hello @syinari0123,

Sorry for the late response.

Yes, the depth is perpendicular to the projection plane.

We've been doing some experiments and we've got results without noticiable distortion.
Your code is correct to me, and the intrinsic matrix shoud be the following one:

    K = [[f, 0, Cu],
         [0, f, Cv],
         [0, 0, 1 ]]

Given a 2D point ```p2d = [u,v,1]```, your world point position ```P = [X,Y,Z]``` will be:

    P = ( inv(K) * p2d ) * depth

We're working on a example integrating the point cloud with the python client.
I hope we are going to release it soon :)

Due to #58 we can't get the correct camera world rotation, but it will be fixed in the next release.

syinari0123, 2017-11-30:
Thank you for replying.
I modified my code following as what you say, I was able to get correct result ! Thank you! :)

I thought depth is the ray length to each point from the pole (like LIDAR sensor's value, what is called, "r in polar coordinate system" ), which caused distortion in my code.

marcgpuig, 2017-11-30:
Glad it worked! :)

closed 2017-11-30 by marcgpuig

syinari0123, 2018-02-08:
Sorry, I have some question about it.
I implemented a code which transforms a point(u,v) on the image into wold coord's point(x,y,z) according to your formula.
I found it has some distortion between images. 
Example of transformation between 2 images(t,t+1) is following.

img(t)
[image]
img(t+1)
[image]
The result of transformation t->t+1 is following.(Points on img(t) are projected on img(t+1).)
[image]

At first glance, you think transformation is good. However, plotting all these points in world coords and comparing these, you will find this has some distortion. Following figure is the result of plotting point clouds. This figure is seen from right side of this world coords.
(Red points are from img(t), Green points are from img(t+1), Blue line is all trajectory from the moving vehicle.)
I think red(img(t)) and green(img(t+1)) would match on this map if there is no distortion.
[image]

I think this distortion is caused from [image] provided by Unreal Engine.
Maybe [image] or [image] effect this distortion for creating realistic image.

Could you tell me how to undistort the image taken by CARLA? Or If you could, please add a function to adjust or turn off this distortion in CARLA simulator in the future...

Best regards.