carla-simulator/carla #58: Send vehicle's roll, pitch, and yaw to client

nsubiron, 2017-11-29:
Currently we are sending the Cartesian orientation of the vehicle. This only provides two angles. Sending roll, pitch, and yaw will provide the full rotation of the vehicle.

Proposed proto message
```protobuf
message Vector3D {
  float x = 1;
  float y = 2;
  float z = 3;
}

message Rotation3D {
  float pitch = 1;
  float yaw = 2;
  float roll = 3;
}

message Transform {
  Vector3D location = 1;
  Vector3D orientation = 2 [deprecated=true];
  Rotation3D rotation = 3;
}
```

NikolausDemmel, 2017-11-29:
It would great if the protocol / documentation would mention, which Euler angle convention is being used (i.e. in which order the rotations are applied and if they are with static axes or rotating axes) and also what rotation is being sent (world-to-cam or cam-to-world, I'm guessing the latter) such that the rotation can be interpreted correctly.

WenchaoDing, 2017-11-30:
Instead of using Euler angles, how about directly using quaternion (w, x, y, z), which is robust and convenient, also matches the ROS convention? Just an advice. :)

nsubiron, 2017-11-30:
Indeed, we are lacking documentation on this. We are already working on it, soon we will provide too a python example to convert a depth image to a point cloud in world coordinates.

For every agent in the scene we send this transform, it defines the agent-to-world transform. We basically send Unreal's [FRotator](https://docs.unrealengine.com/latest/INT/API/Runtime/Core/Math/FRotator/index.html) data. Unreal's documentation is scarce, but going through their code I can guess that their rotation is applied in the order Roll (clockwise), then Pitch (ccw), then Yaw (ccw). Since Unreal has left-handed axis this complicates things too, the magic rotation matrices I deduced are

```py
import numpy as np

# ...

transform = measurements.player_measurements.transform

yaw = np.radians(transform.rotation.yaw)
roll = np.radians(transform.rotation.roll)
pitch = np.radians(transform.rotation.pitch)

yaw_matrix = np.matrix([
    [math.cos(yaw), -math.sin(yaw), 0],
    [math.sin(yaw), math.cos(yaw), 0],
    [0, 0, 1]
])

pitch_matrix = np.matrix([
    [math.cos(pitch), 0, -math.sin(pitch)],
    [0, 1, 0],
    [math.sin(pitch), 0, math.cos(pitch)]
])

roll_matrix = np.matrix([
    [1, 0, 0],
    [0, math.cos(roll), math.sin(roll)],
    [0, -math.sin(roll), math.cos(roll)]
])

rotation_matrix = yaw_matrix * pitch_matrix * roll_matrix

translation = np.matrix([
    [transform.location.x],
    [transform.location.y],
    [transform.location.z]
])

camera_rel_position = np.matrix([[x], [y], [z]])

camera_world_position = (rotation_matrix * camera_rel_position) + translation
```

In the case of the cameras we don't send the transform of each camera, but the transform of the player vehicle. Since the transform of the camera-player is set when defining the cameras, it is possible to get world-camera transform.

@WenchaoDing, seeing how arbitrary Unreal's convention is we may consider using quaternions instead; we'll think about this option.

NikolausDemmel, 2017-11-30:
Quaternions also have different conventions, but I agree that they might be a better choice, in particular if you anticipate people using the output with ROS.

For quaternion conventions, see http://www.iri.upc.edu/people/jsola/JoanSola/objectes/notes/kinematics.pdf. I would probably make sense to use what ROS (and other frameworks like Eigen, Ceres, ...) is using, which is "Hamilton".

For ROS documentation about their quaternion use, see http://www.ros.org/reps/rep-0103.html, and maybe http://wiki.ros.org/Tutorials/Quaternions is also useful.

NikolausDemmel, 2017-11-30:
@nsubiron, I see its a bit of a mess. But once you have the rotation matrix for right-handed coordinate system, you should be fine.

For conversions into other representations, this single-file library https://www.lfd.uci.edu/~gohlke/code/transformations.py.html is nice. It's also part of ROS and  used in the "Quaternion Tutorial" above.

closed 2018-01-23 by nsubiron

KitKat7, 2018-10-26:
@nsubiron, I find the transform matrix in your comment is a little different from the one in [carla_transform_to_ros_transform](https://github.com/carla-simulator/carla/blob/master/carla_ros_bridge/src/carla_ros_bridge/transforms.py) within [carla_ros_bridge](https://github.com/carla-simulator/carla/tree/5fb2dcdba1456c265bddf2052d5e31de80030822/carla_ros_bridge). If my understanding is correct, the transform might be from
```python
# ...
yaw_matrix = np.matrix([
    [math.cos(yaw), math.sin(yaw), 0],
    [-math.sin(yaw), math.cos(yaw), 0],
    [0, 0, 1]
])

pitch_matrix = np.matrix([
    [math.cos(pitch), 0, math.sin(pitch)],
    [0, 1, 0],
    [-math.sin(pitch), 0, math.cos(pitch)]
])

roll_matrix = np.matrix([
    [1, 0, 0],
    [0, math.cos(roll), math.sin(roll)],
    [0, -math.sin(roll), math.cos(roll)]
])
# ...
camera_rel_position = np.matrix([[x], [-y], [z]])
# ...
```

> Indeed, we are lacking documentation on this. We are already working on it, soon we will provide too a python example to convert a depth image to a point cloud in world coordinates.
> 
> For every agent in the scene we send this transform, it defines the agent-to-world transform. We basically send Unreal's [FRotator](https://docs.unrealengine.com/latest/INT/API/Runtime/Core/Math/FRotator/index.html) data. Unreal's documentation is scarce, but going through their code I can guess that their rotation is applied in the order Roll (clockwise), then Pitch (ccw), then Yaw (ccw). Since Unreal has left-handed axis this complicates things too, the magic rotation matrices I deduced are
> 
> ```python
> import numpy as np
> 
> # ...
> 
> transform = measurements.player_measurements.transform
> 
> yaw = np.radians(transform.rotation.yaw)
> roll = np.radians(transform.rotation.roll)
> pitch = np.radians(transform.rotation.pitch)
> 
> yaw_matrix = np.matrix([
>     [math.cos(yaw), -math.sin(yaw), 0],
>     [math.sin(yaw), math.cos(yaw), 0],
>     [0, 0, 1]
> ])
> 
> pitch_matrix = np.matrix([
>     [math.cos(pitch), 0, -math.sin(pitch)],
>     [0, 1, 0],
>     [math.sin(pitch), 0, math.cos(pitch)]
> ])
> 
> roll_matrix = np.matrix([
>     [1, 0, 0],
>     [0, math.cos(roll), math.sin(roll)],
>     [0, -math.sin(roll), math.cos(roll)]
> ])
> 
> rotation_matrix = yaw_matrix * pitch_matrix * roll_matrix
> 
> translation = np.matrix([
>     [transform.location.x],
>     [transform.location.y],
>     [transform.location.z]
> ])
> 
> camera_rel_position = np.matrix([[x], [y], [z]])
> 
> camera_world_position = (rotation_matrix * camera_rel_position) + translation
> ```
> 
> In the case of the cameras we don't send the transform of each camera, but the transform of the player vehicle. Since the transform of the camera-player is set when defining the cameras, it is possible to get world-camera transform.

commit d2b46bad (2017-11-29): Fix #58 Send vehicle's roll, pitch, and yaw to client