carla-simulator/carla #59: Multiple cameras independent from the vehicle

ndrplz, 2017-11-29:
Hi guys, 

first of all thank you for your wonderful work!

I was wondering if there's any chance to use different cameras which are positioned in a certain real-world point and are independent from the position of the vehicle.

I started diving into the carla's code just today, so I'm certain I still miss something. Up to now I found that the Camera object [has a set position method](https://github.com/carla-simulator/carla/blob/e206383ac4d2c87ed28267021ea20f37c9d3ca36/PythonClient/carla/sensor.py#L55-L58). However, in the `client_example.py` it [looks like this position is always relative to the car position](https://github.com/carla-simulator/carla/blob/e206383ac4d2c87ed28267021ea20f37c9d3ca36/PythonClient/client_example.py#L64).

Conversely, I'd like to have the possibility to set one or more cameras in precise town position, independently from where the player's vehicle currently is.

Thank you again for your time
Andrea

nsubiron, 2017-11-29:
Hi Andrea,

Exactly, from the client side it is only possible to add cameras relative to the player vehicle. I will mark this as "feature request" because I think it can be an useful feature for more people.

Now, if you are willing to hack into the Unreal project, you can already save images to disk from an arbitrary position. We have an actor called "Scene Capture To Disk Camera", just add this actor to the scene from the Modes tab and drop it wherever you want in the map

[image]

Now in the details panel you can adjust the camera parameters as you would do in the Python client

[image]

If you hit Play (or Simulate), this cameras will start recording images to disk. They are pretty slow but do the job.

ndrplz, 2017-11-30:
Thank you for your prompt response.

Indeed I guess that this feature would be useful for many applications! In the meanwhile I'll give a try to the hack you suggested.

Thanks,
A.

rawalkhirodkar, 2018-04-04:
Thank you to the Carla Team, great work!

@nsubiron I tried the above solution to assemble two cameras (rgb and semantic segmentation). 

The sensor id for both of the cameras is 0 by default, and it seems I cannot change it in the editor (the field is grayed out).

The captured images are correct when the scene contains only one of the camera (cam_rgb or cam_segmentation). I get weird images when both the cameras are in the scene. 

Is this because they have the same sensor id? Please suggest a solution. Thanks.

nsubiron, 2018-04-05:
Hi @rawalkhirodkar,

I'm getting the same error, this seems to be a bug introduced recently. We are investigating why this happens but it doesn't seem to be related to the sensor id.

I added the issue #334.

rawalkhirodkar, 2018-04-05:
Thank you for the response @nsubiron 
Could you redirect me to the latest version which might be free of this bug?

nsubiron, 2018-04-06:
@rawalkhirodkar Yes, I'll let you know when we have a fix

juaxix, 2018-04-12:
@rawalkhirodkar this bug is fixed in #333 
cheers

rawalkhirodkar, 2018-04-12:
@juaxix It works. Thank you!

krydm, 2018-06-04:
Hello, I'm looking for the very same feature, but not sure whether I can use it as it is. What I need is to set several SceneCaptureToDisk cameras and capture frames from them, but at the same time I need to get their sensor data, i.e., have to run something like(from carla docs):measurements, sensor_data = client.read_data()

measurements, sensor_data = client.read_data()
for agent in measurements.non_player_agents:
    agent.id # unique id of the agent
    if agent.HasField('vehicle'):
        agent.vehicle.forward_speed
        agent.vehicle.transform
        agent.vehicle.bounding_box

This code works fine for cameras added through settings.add_sensor()
However, if I just add SceneCaptureToDiskCameras through the editor, they, naturally, do not appear in the sensor_data.keys(). Is there any way to create them through the client code(or to refer to these editor-created cameras in the client code) in order to pass them to add_sensor(), in the same way as we do with cameras in client_example.py?

Thanks a lot

nsubiron, 2018-08-01:
This is already possible with API 0.9.0.

closed 2018-08-01 by nsubiron

rishimadhok, 2018-10-09:
Is it possible to change the focal length of these static sensors?