carla-simulator/carla #60: Optical flow and camera pose data?

ZheC, 2017-11-30:
First of all, thank you for open-source this great project! It is really helpful!

Could you provide the code to extract the ground-truth optical flow and camera extrinsic from the simulator as well? This will be very helpful to many computer vision research problems.

nsubiron, 2017-11-30:
Regarding the camera extrinsic, it will be ready in the next release, take a look at issue #58

chaene, 2017-12-01:
Thanks for considering this feature request.

For our research it would be really helpful to have optical flow and I believe this would not only be useful to us but for many other research groups. Currently there are almost no datasets available which have depth and flow at the same time. However, there is a growing interest in learning 3D reconstruction and motion.

Also, we are happy to help with implementing this if it would be helpful.

nsubiron, 2017-12-05:
The ideal way of implementing this would be creating an Unreal material, like we do now with depth and semantic segmentation ("Unreal/CarlaUE4/Plugins/Carla/Content/PostProcessingMaterials/"). However we aren't certain this can be done this way. Perhaps checking how the motion blur is done may give some clues.

infoclogged, 2018-10-13:
Ground truth Optical Flow can be done offline if you know the displacement of the vehicle. I do it all the time. I am not sure, if it can be a part of a virtual dataset generator. I think it does not belong here, otherwise, all the CV algorithms such as Harris, ground truth needs to be provided. Happy to know what the community thinks about this.

lijecaru, 2019-02-13:
Any updates on computing optical flow in Carla? The latest release has a lot of updates, I'm just not sure if there's a clearer way to do it now.

germanros1987, 2020-03-19:
@marcgpuig @doterop for visibility:

Let's try to have optical flow GT for 0.9.10 (July?)

magehrig, 2021-08-06:
Note that the feature has been implemented as of 0.9.12: https://carla.readthedocs.io/en/0.9.12/ref_sensors/#optical-flow-camera.

closed 2024-05-10 by MarcelPiNacy-CVC