carla-simulator/carla #61: Carla UE4 is lagging

researchlabs, 2017-11-30:
Why Carla UE4 is lagging?

Command: ./CarlaUE4.sh /Game/Maps/Town02 -fps=15 -ResX=800 -ResY=600 -windowed -carla-no-networking -carla-settings=./CarlaSettings.ini

I've recorded 45 second. We can see video pulls on 
https://youtu.be/WSk0B4zuOxc
How I can make simulator more convenient for work?

system info:
OS: Ubunty 16.04.3 LTS
CPU: 4x i3-4130
RAM: 16Gb

nsubiron, 2017-11-30:
Which GPU you have? That's what mostly affects Carla speed. If you are running without networking, check that you are not activating any extra cameras in the CarlaSettings.ini, that also slows down Carla significantly.

And a detail, `-fps=15` has no effect unless you provide `-benchmark` too.

researchlabs, 2017-11-30:
GPU:
GK208 [GeForce GT 720]
driver: nvidia_384
64bit/33MHz
Mem: 2Gb

Camera:
I have default config in camera section, there is  one section 
[CARLA/SceneCapture/MyCamera]
PostProcessing=SceneFinal

-benchmark:
I don't get what command to check benchmark.
is it ./CarlaUE4.sh -benchmark or GPU benchmark.

What level of GPU would you recommend?

....

felipecode, 2017-11-30:
Hey @researchlabs,
Unfortunately this lag you are having is probably normal considering your graphics hardware.
CARLA is currently computationally demanding for video boards. We perform most of our test
in high end nvidia-pascal architecture gpus and also on the best ones from nvidia-maxwell architecture.

Some benchmarking for other gpus would be valuable.

Thanks for your interest !

nsubiron, 2017-11-30:
I see, GT 720 is indeed a bit too low for CARLA, unfortunately CARLA is very demanding, we are trying to improve this in the future. Take a look at #10.

To disable the camera it's enough to set the settings "Cameras" empty or commented-out in the CarlaSettings.ini
```ini
[CARLA/SceneCapture]
Cameras=
```

The benchmark param is used the Carla executable, it forces Carla to run at fixed-frame rate

    $ ./CarlaUE4.sh -benchmark -fps=15

closed 2017-12-01 by felipecode

Sultan91, 2023-05-09:
I have geforce 1060 6gb laptop. Is this also an issue for me?