carla-simulator/carla #63: Non-player agents info for objects within the FoV of the camera only

suvam-bag, 2017-12-01:
Is there a way to get the ground truth of the objects only within the FoV of the camera?

felipecode, 2017-12-01:
Hello @suvam-bag  !
At this moment we don't have such function. However, i think this could
be quite useful ! 
 From the client side, It could  be implemented from the data we already have.
The client receives the camera pose , FoV and the positions of all the objects
present in the scene. With this, one could compute a function that projects 
the camera view on the 3D space and get all the visible objects.

This is also possible to be done from server side. The rendering engine already
needs to get the visible objects, so this should be low cost. One could investigate how to do that with unreal.

Currently we are working on other things. Help from the community would be useful!

suvam-bag, 2017-12-01:
Thanks @felipecode for the detailed answer! I will try to create a mapping function between the image pixel and the position of the ego-vehicle in the client side.

YashBansod, 2018-02-22:
@suvam-bag. Were you able to create the mapping? If you did, could you help me with it?

YashBansod, 2018-02-27:
@felipecode am I correct in understanding that the code to generate semantic segmentation images is written by the Carla team. If so, is there some way for a simple tweak in that algorithm to create instance segmentation?

I basically want to test a Faster-RCNN for object detection in Carla. Now to get the ground truth bounding boxes I can have 2 approaches:

1. Somehow using the given camera data and the 3D position, orientation and size of the object, I project the bounding box onto my camera view. But this requires some knowledge of perspective mapping and other CV topics. Still, it wont really work well since the Z axis length of the bounding box is not correct in the given data.

2. Use the Semantic Segmentation to create a bounding box for each type of class in the scene. This is really easy to implement where your topX=minX , topY = minY, W=(maxX - minX) and H=(maxY - minY) but you should have only one instance of the class in the scene for this kind of bounding box creation to be done. This would work easily if we were provided with instance segmented images instead.

Anyway, If you could suggest me on a way I could create the bounding box of the objects in camera FOV, it would really be helpful.

YashBansod, 2018-02-28:
@nsubiron  @felipecode , are there any plans to enable instance segmentation?

I was able to get the bounding boxes as long as the pixels of the instances of a class are not intersecting. But I still can't get bounding boxes correctly if the pixels are intersecting.

JieMEI1994, 2018-06-01:
@felipecode @nsubiron  Thanks for your amazing carla project.
@YashBansod  @suvam-bag Thanks your question. And could you share some experience in the creating the mapping function?

I have met the same problem in transferring the 3d info of the non-player into 2d info in a camera, which will be used to evaluate the object tracking algorithm. From the server side, the position info. and bounding box info. of the player, the non-players and the camera can be easily gotten in both general coordinate and player-relative coordinate. Still the problem is how to mapping the 3d info. into the 2d info. in the camera.

what we know:
 * camera => FOV, image_size, position, rotation
 * non-player => transform info. related to general coordinate, bounding_box info. related to own coordinate
 * player => transform info. related to general coordinate
what we want : 
the ground truth info. calculated according to carla environment.

@YashBansod you have mentioned two ways and you also said that you have been able to get the bbox with the segmentation info. would you mind sharing more info. of your idea? 
I have been working on the first way and found it difficult to get the mapping function since the perspective problem.

YaduKini, 2019-02-18:
Hi @mhusseinsh,

I am trying to solve the same problem. Did you find a solution to it?
please let me know?_

Wh1teSh4dow, 2019-03-30:
> Hello @suvam-bag !
> At this moment we don't have such function. However, i think this could
> be quite useful !
> From the client side, It could be implemented from the data we already have.
> The client [...]

hi @felipecode, I'm deeply interested in what you're specifying above. Could you please elaborate on how this is done? Thank you!

germanros1987, 2020-03-19:
Problem solved in current versions of CARLA.

closed 2020-03-19 by germanros1987

cinjon, 2021-03-20:
@germanros1987 where do you see that this was solved?