carla-simulator/carla #64: How to transform box_extent to boundingbox

johnlanbor, 2017-12-01:
Hi,carla team
how can i transform the box_extent(veichels and pedstrain) to bounding box in 2D image:
non_player_agents {
  id: 1093932246
  vehicle {
    transform {
      location {
        x: 13203.0
        y: 20117.0
        z: 137.0
      }
      orientation {
        x: 7.152557373046875e-07
        y: 1.0
      }
    }
    box_extent {
      x: 230.0
      y: 108.0
      z: 32.0
    }
  }
}

nsubiron, 2017-12-01:
Hi @johnlanbor,

The transform defines the location and orientation of the agent. The bounding box is assumed to be centred at the agent, i.e., same transform. The box extent gives the radii dimensions of the box.

In that case, there is a vehicle centered at location (x=13203cm, y=20117cm, z=137cm), oriented towards Y axis, it's 460cm long and 216cm wide. The height of the vehicle is currently not set in the bounding box, see https://github.com/carla-simulator/carla/issues/26#issuecomment-347810323.

TheNihilisticRobot, 2017-12-01:
[image]

@johnlanbor 
Here is a little Image displaying this info. Soon It will also be in the documentation.

closed 2017-12-01 by TheNihilisticRobot