carla-simulator/carla #65: Changing map from the client side.

felipecode, 2017-12-01:
Right now the town has to be loaded previously when you start
the server, e.g:
_sh CarlaUE4.sh /Game/Maps/Town02_

When you call for episode restarts on the client side, it is not possible
to change the town, for that you need to manually restart  the server
and select another town.
This reduces the autonomy of training, testing and benchmarking.

I suggest another line on the settings file.
```
[CARLA/Server]
Map= Town02
```
The server should detect a change in the map and do a full reload.

juaxix, 2018-05-22:
Did you test the [level_transition](https://github.com/carla-simulator/carla/tree/level_transition) branch already @felipecode ?

felipecode, 2018-06-04:
@juaxix 
Check my PR #472

juaxix, 2018-06-07:
It was merged and you can try now, with this PR #482

felipecode, 2018-06-16:
Still have some errors:
Check my comment on the PR
[2018.06.16-14.53.42:497][ 91]LogCarla: Error: The map in CarlaSettings is not the current one: Town02 !

felipecode, 2018-06-17:
It passes the test, but after some time running , after changing the town. It crashes. with error above

juaxix, 2018-06-17:
I see, this is a problem with the [initgame](https://github.com/carla-simulator/carla/blob/level_transition/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/Game/CarlaGameModeBase.cpp#L75) function then... needs some test + research

closed 2018-08-23 by stale[bot]

Danmou, 2019-02-19:
Is this still on the roadmap? It would be nice to be able to change map from the client without restarting the server.

nsubiron, 2019-02-19:
Yup, it's almost done in #1272.