carla-simulator/carla #66: Fuse -> Makehuman

L0rdCha0s, 2017-12-04:
Hi there,

We have performed some work to use 'makehuman' characters in place of Adobe Fuse, thus allowing redistribution in the open source release.

Would the project maintainers be interested in incorporating our changes, on the assumption that we get corporate signoff on our side?

nsubiron, 2017-12-05:
Hi @L0rdCha0s,

Thanks for being willing to share your pedestrians with the community! 

We are currently evaluating different options to solve the license issue with our pedestrians, so we don't know for sure yet how we will proceed with this. We will let you know soon if we are interested in incorporating your changes into the project.

xie-dongping, 2018-01-16:
@nsubiron 
MakeHuman has an exception clause for [exported model](http://www.makehuman.org/license_explanation.php) if one uses the official release. In this case, the exported model are licensed through CC0 (almost like public domain) instead of the restrictive AGPL according to them.

nsubiron, 2018-02-02:
We finally decided to work on our own pedestrian models to avoid having any more license issues. They will be released in the coming weeks, please track the progress at #188.

Nevertheless, we are happy to add a link to external alternative pedestrians if anyone is willing to publicly share them.

closed 2018-02-02 by nsubiron