carla-simulator/carla #67: Some question about Autopilot control mode

syinari0123, 2017-12-04:
Hi, I have a question.
I'm measuring the data from driving car under autopilot_control. But I found autopilot_control_car sometimes drive recklessly. Please tell me does this case often happen in your environment, too?

In addition, I want to analyze this bag, so, I also want to know "how autopilot_control decides the way to act." I infer this is the result of reinforcement learning of your paper's experiment, which cause some reckless driving...

nsubiron, 2017-12-04:
Hi @syinari0123,
Yes, exactly. You can find the autopilot code [here](https://github.com/carla-simulator/carla/blob/master/Unreal/CarlaUE4/Plugins/Carla/Source/Carla/AI/WheeledVehicleAIController.cpp). It is the same AI that use all the other non-player vehicles to drive too.

What I'm guessing is happening is that you have the synchronous mode disabled and your client code is slower than the server. If that's the case, the server should be printing _"Warning: No control received from the client this frame!"_. In asynchronous mode, the server won't wait for the control message, so if the client sends the autopilot control back too late, it will be applied to the vehicle some frames later than desired. This can cause the car to brake or steer too late.

syinari0123, 2017-12-04:
Thank you for your prompt response!
As your guessing, I found "Warning: No control received from the client this frame!" in server printing. I changed synchronous mode enable, which made my autopilot-control-car drive carefully! :)
Thanks!

closed 2017-12-04 by syinari0123