carla-simulator/carla #68: Lidar support

parilo, 2017-12-04:
Hi, All.
I've added Lidar support and now cleaning my code and fixing minor problems. Soon I will be able to make cross repo pull request. Will my code be useful for Carla project?

What is supported:
- adding any number of lidars through configuration ini file
- for individual lidars set number of lasers and vertical angles, number of points per second, position, orientation
- currently lidar returns XYZ coordinates in lidar frame
- points are serialized and available in python client and visualized in ./carla_manual_control.py

felipecode, 2017-12-04:
Hey Parilo !
Thank you a lot for your contribution ! We are definitely looking into your code and trying to add it to the release 0.7.2.

parilo, 2017-12-05:
I've created pull request with my changes. I've updated the code with recent master. Now it is ready for review.

nsubiron, 2017-12-05:
Thanks @parilo! We might take a couple of weeks to review it since now we are very busy, but we are definitively interested.

PR #105 (2018-03-01): Adding 360 Lidars support
Adding the changes of #69.

Resolve #68.

- - -
[image]

You may add any number of lidars to the car. Lidars have following settings with specified default values:

```
# Number of lasers
Channels = 32

# Measure distance
Range = 5000

# Points generated by all lasers per second
PointsPerSecond = 100000

# Lidar rotation frequency
RotationFrequency = 10

# Upper laser angle, counts from horizontal,
# positive values means above horizontal line
UpperFovLimit = 10

# Lower laser angle, counts from horizontal,
# negative values means under horizontal line
LowerFovLimit = [... cut]
closed 2018-03-01 by nsubiron