carla-simulator/carla #71: Some questiones about multi-cars controller .

18605973470, 2017-12-08:
I want to train my autonomous drving model under a specific scenario in which a car follows a vehicle platoon. 
And I have to construct this scenario.
So I want to know if it is easy to control multi-cars in carla. ( The last one is controlled by trained model and   the platoon is controlled by a simple program whose aim is going straight forward.  

Any suggestions would be appreciated !

felipecode, 2017-12-08:
Hello @18605973470 .
Seems that you are working on a very nice scenario!
Unfortunately CARLA still does not allow controlling multiple vehicles directly
in a way that would be easy to specify a scenario.
There is one player vehicle and the others are controlled by an algorithm
programmed on the server side.  Basically they just drive around, obeying the traffic rules.
You can either try to program this behavior on server side, or wait our development,
since, adding defined scenarios is in our roadmap! 

Cheers !

closed 2017-12-20 by nsubiron

ngeley, 2018-02-21:
Can you point me to a resource to program controllers for the vehicle on the server side? 
Also I'm not sure how that is different than programming on the client side. Thank you!

shubhamck, 2019-03-21:
> Can you point me to where I could program the controllers for the vehicles on the server side and use this to create multiple "ambient"/environment vehicles (which can affect the agent vehicle)?

Did you figure this out?

aadityasp1, 2019-07-22:
@ngeley Were you able to control multiple cars ?

stavBodik, 2019-11-24:
Is this same this days ? there is no option to control multiple vehicles? I would like to give a special behavior for each of the cars in the scene ?

AOOOOOA, 2022-06-01:
> Hello @18605973470 . Seems that you are working on a very nice scenario! Unfortunately CARLA still does not allow controlling multiple vehicles directly in a way that would be easy to specify a [...]

Hi @felipecode ,

Do you guys have any update on this problem in the latest version CARLA? I mean, control multiple vehicles in one client program.

Thanks!