carla-simulator/carla #72: How are semantic segmentation colors picked?

xevo-vstojkovic, 2017-12-09:
I have been looking at the source to understand how semantic segmentation works. I've been able to figure out most of the process so far:

1. Each static mesh for each actor is tagged based on its path.
1. When a mesh is tagged, its custom depth stencil value is set to one of the `ECityObjectLabel` values and it's allowed to render in the custom depth buffer.
1. The semantic segmentation camera is set up to use a post processing blendable material `GTMaterial` defined in the Carla plugin.
1. The material uses the custom stencil value to determine the emissive color of the material.

Here's what I don't understand: `GTMaterial` normalizes the custom stencil value, multiplies it with a constant red color and uses that as the emissive color, yet the final result has a variety of hues (reds, greens, purples, blues, etc.)

What am I missing? How are these final colors determined?

marcgpuig, 2017-12-09:
Hi @xevo-vstojkovic,

Check out the [Cameras and sensors](http://carla.readthedocs.io/en/latest/cameras_and_sensors/#semantic-segmentation) documentation :)
The server provides an image with the tag information encoded in the **red channel**.

This coloring process is done in the **client side** with [```labels_to_cityscapes_palette```](https://github.com/carla-simulator/carla/blob/9855fe35f9691a4e9346f6a4525de08777c9cbf9/PythonClient/carla/image_converter.py#L52-L76) function.

closed 2017-12-21 by marcgpuig

hrdisaac, 2018-05-11:
One problem of using stencil buffer for segmentation is that it limits the number of tags to be no more than 256. If you want to have more tags for segmentation, you can't use stencil buffer for this purpose. Any suggestion for this?

marcgpuig, 2018-05-11:
Hi @hrdisaac,

That's one of our main problems right now. Sorry for that.

This is a big limitation and does not allow us to generate instance semantic segmentation and has a performance impact too.
We are waiting to Unreal (or someone) to give us an elegant solution.

Maybe modifying the engine itself, you can do something like generate a custom buffer with this information. We are trying to avoid any engine modification because it may generate problems with some future Unreal updates or with onthers parts of the engine.

Regards.

hrdisaac, 2018-05-22:
In order to support more than 256 tags without modifying the engine itself, is it possible to dynamically create a transparent material per static mesh and set a custom parameter on that material and somehow access that custom parameter in the post process pass? Or any way to make use of postprocessinput3 to postprocessinput6 (marked as not usually used by Unreal) to inject the tags? Or create two render targets to render the same material for twice, so in the shader it can use one custom stencil value as the lower bits and the other as the higher bits, then combine them in the shader to render the final color for segmentation?

marcgpuig, 2018-05-24:
Hey @hrdisaac,
I've answered your question in the other issue #76 :)

leimao, 2018-11-07:
This thread helped me.