carla-simulator/carla #73: Segmentation fault (core dumped)

coolKeen, 2017-12-10:
smielab@smielab-Z97X-UD3H:~/simulator/CARLA_0.7.0$ ./CarlaUE4.sh /Game/Maps/Town01 -carla-server -benchmark -fps=15 -windowed -ResX=800 -ResY=600 -opengl3

Hi,everyone~
When I download the release package and run the command above,I get the information below.Do every know what's happened?Thanks a lot~
(Computer setting:
cpu:i7-4790
memory:30GB
gpu:GeForce GTX TITAN X(12GB)
)

Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogInit: Display: RandInit(0) SRandInit(0).
LogTaskGraph: Started task graph with 5 named threads and 14 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.054507
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin Carla
LogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 
LogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
LogInit: Build: ++UE4+Release-4.17-CL-0
LogInit: Engine Version: 4.17.2-0+++UE4+Release-4.17
LogInit: Compatible Engine Version: 4.17.0-0+++UE4+Release-4.17
LogInit: Net CL: 0
LogDevObjectVersion: Number of dev versions registered: 16
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0
LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 2
LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 20
LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 28
[... 501 lines cut ...]
[2017.12.10-02.20.26:652][  0]LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
[2017.12.10-02.20.26:652][  0]LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
[2017.12.10-02.20.26:653][  0]LogInit:  CurlRequestOptions (configurable via config and command line):
[2017.12.10-02.20.26:653][  0]LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
[2017.12.10-02.20.26:653][  0]LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
[2017.12.10-02.20.26:653][  0]LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
[2017.12.10-02.20.26:653][  0]LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
[2017.12.10-02.20.26:653][  0]CrashReportClientLog: Sending HTTP request: https://datarouter.ol.epicgames.com/datarouter/api/v1/public/data?AppID=CrashReporter&AppVersion=4.17.2-0%2B%2B%2BUE4%2BRelease-4.17&AppEnvironment=Release&UploadType=crashreports&UserID=00024567230c0f09001b03776a830512%7C%7Ceed0a2ca7cf9428db8c9c1407572504d
[2017.12.10-02.20.26:716][  0]LogHttp: Warning: 0x7f9cfe0ad300: request failed, libcurl error: 7 (Couldn't connect to server)
[2017.12.10-02.20.26:716][  0]CrashReportClientLog: OnProcessRequestComplete(), State=SendingFiles bSucceeded=0
[2017.12.10-02.20.26:716][  0]CrashReportClientLog: Warning: File upload failed to data router
[2017.12.10-02.20.26:716][  0]CrashReportClientLog: State change from SendingFiles to Cancelled
[2017.12.10-02.20.27:651][  0]LogLinux: FLinuxPlatformMisc::RequestExit(0)
[2017.12.10-02.20.27:651][  0]LogGenericPlatformMisc: FPlatformMisc::RequestExit(0)
[2017.12.10-02.20.27:685][  0]CrashReportClientLog: Final state (Receiver) = Cancelled
[2017.12.10-02.20.27:685][  0]CrashReportClientLog: Final state (Receiver) = Unknown UploadState value
[2017.12.10-02.20.27:685][  0]LogExit: Preparing to exit.
[2017.12.10-02.20.27:685][  0]LogExit: Object subsystem successfully closed.
[2017.12.10-02.20.27:686][  0]LogModuleManager: Shutting down and abandoning module HTTP (8)
[2017.12.10-02.20.27:691][  0]LogModuleManager: Shutting down and abandoning module SSL (7)
[2017.12.10-02.20.27:691][  0]LogModuleManager: Shutting down and abandoning module CrashDebugHelper (4)
[2017.12.10-02.20.27:691][  0]LogModuleManager: Shutting down and abandoning module CoreUObject (2)
[2017.12.10-02.20.27:691][  0]LogExit: Exiting.
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)

nhynes, 2017-12-10:
It might be because you're using `-opengl3`. I tried the same thing in a VM and it summarily crashed.

coolKeen, 2017-12-10:
But the problem is that if I don't use -opengl3,it give me the information below which say my driver don't support opengl4.3.
My driver is installed from "NVIDIA-Linux-x86_64-384.98.run".I think it is the newest and sholuld support opengl4.3.
I really don't know how to do:(.

smielab@smielab-Z97X-UD3H:~/simulator/carla_0.7.0$ ./CarlaUE4.sh
Increasing per-process limit of core file size to infinity.
LogPlatformFile: Using cached read wrapper
LogInit: Display: RandInit(35190807) SRandInit(35190807).
LogTaskGraph: Started task graph with 5 named threads and 14 total threads with 3 sets of task threads.
LogStats: Stats thread started at 0.640302
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin Carla
LogFileManager: Warning: ReadFile failed: Count=0 Length=4096 Error=errno=21 (Is a directory) for file 
LogInit: Using libcurl 7.48.0-DEV
LogInit:  - built for x86_64-unknown-linux-gnu
LogInit:  - supports SSL with OpenSSL/1.0.2h
LogInit:  - supports HTTP deflate (compression) using libz 1.2.8
LogInit:  - other features:
LogInit:      CURL_VERSION_SSL
LogInit:      CURL_VERSION_LIBZ
LogInit:      CURL_VERSION_IPV6
LogInit:      CURL_VERSION_ASYNCHDNS
LogInit:      CURL_VERSION_LARGEFILE
LogInit:      CURL_VERSION_TLSAUTH_SRP
LogInit:  Libcurl: checking if '/etc/pki/tls/certs/ca-bundle.crt' exists
LogInit:  Libcurl: checking if '/etc/ssl/certs/ca-certificates.crt' exists
LogInit:  CurlRequestOptions (configurable via config and command line):
LogInit:  - bVerifyPeer = true  - Libcurl will verify peer certificate
LogInit:  - bUseHttpProxy = false  - Libcurl will NOT use HTTP proxy
LogInit:  - bDontReuseConnections = false  - Libcurl will reuse connections
LogInit:  - CertBundlePath = /etc/ssl/certs/ca-certificates.crt  - Libcurl will set CURLOPT_CAINFO to it
LogInit: Build: ++UE4+Release-4.17-CL-0
LogInit: Engine Version: 4.17.2-0+++UE4+Release-4.17
LogInit: Compatible Engine Version: 4.17.0-0+++UE4+Release-4.17
LogInit: Net CL: 0
LogDevObjectVersion: Number of dev versions registered: 16
LogDevObjectVersion:   Dev-Blueprints (B0D832E4-1F89-4F0D-ACCF-7EB736FD4AA2): 10
LogDevObjectVersion:   Dev-Build (E1C64328-A22C-4D53-A36C-8E866417BD8C): 0
LogDevObjectVersion:   Dev-Core (375EC13C-06E4-48FB-B500-84F0262A717E): 2
LogDevObjectVersion:   Dev-Editor (E4B068ED-F494-42E9-A231-DA0B2E46BB41): 20
LogDevObjectVersion:   Dev-Framework (CFFC743F-43B0-4480-9391-14DF171D2073): 28
LogDevObjectVersion:   Dev-Mobile (B02B49B5-BB20-44E9-A304-32B752E40360): 0
LogDevObjectVersion:   Dev-Networking (A4E4105C-59A1-49B5-A7C5-40C4547EDFEE): 0
LogDevObjectVersion:   Dev-Online (39C831C9-5AE6-47DC-9A44-9C173E1C8E7C): 0
LogDevObjectVersion:   Dev-Physics (78F01B33-EBEA-4F98-B9B4-84EACCB95AA2): 0
LogDevObjectVersion:   Dev-Platform (6631380F-2D4D-43E0-8009-CF276956A95A): 0
[... 47 lines cut ...]
[2017.12.10-23.43.05:178][  0]LogInit:  - Number of logical cores available for the process: 8
[2017.12.10-23.43.05:178][  0]LogInit:  - Cache line size: 64
[2017.12.10-23.43.05:178][  0]LogInit:  - Memory allocator used: binned2
[2017.12.10-23.43.05:178][  0]LogLinux: Benchmarking clocks:
[2017.12.10-23.43.05:178][  0]LogLinux:  - CLOCK_REALTIME (id=0) can sustain 62648617 (62649K, 63M) calls per second without zero deltas.
[2017.12.10-23.43.05:178][  0]LogLinux:  - CLOCK_MONOTONIC (id=1) can sustain 63137303 (63137K, 63M) calls per second without zero deltas.
[2017.12.10-23.43.05:178][  0]LogLinux:  - CLOCK_MONOTONIC_RAW (id=4) can sustain 14736322 (14736K, 15M) calls per second without zero deltas.
[2017.12.10-23.43.05:178][  0]LogLinux:  - CLOCK_MONOTONIC_COARSE (id=6) can sustain 213978136 (213978K, 214M) calls per second with 99.999879% zero deltas.
[2017.12.10-23.43.05:178][  0]LogLinux: Selected clock_id 1 (CLOCK_MONOTONIC) since it is the fastest support clock without zero deltas.
[2017.12.10-23.43.05:178][  0]LogInit: Linux-specific commandline switches:
[2017.12.10-23.43.05:178][  0]LogInit:  -nodwarf (currently OFF): suppress parsing of DWARF debug info (callstacks will be generated faster, but won't have line numbers)
[2017.12.10-23.43.05:178][  0]LogInit:  -ansimalloc - use malloc()/free() from libc (useful for tools like valgrind and electric fence)
[... cut ...]
[2017.12.10-23.43.05:293][  0]LogInit: Error: _PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'
[2017.12.10-23.43.05:293][  0]LogRHI: Error: OpenGL 4.3 not supported by driver
[2017.12.10-23.43.05:295][  0]LogInit: Error: _PlatformCreateOpenGLContextCore - Could not create OpenGL 4.3 context, SDL error: 'Could not create GL context: GLXBadFBConfig'
[2017.12.10-23.43.46:071][  0]LogLinux: FLinuxPlatformMisc::RequestExit(1)

nsubiron, 2017-12-11:
It seems like an issue with your drivers installation. You may try reinstalling the nvidia drivers. Also, this is a common issue with Unreal Engine, you may try searching in their forums for your specific case.

coolKeen, 2017-12-13:
Thanks for respond~ You are right,I didn't well install graphical driver.After I reinstall the graphical driver correctly, I works!

nsubiron, 2017-12-13:
Great :)

closed 2017-12-13 by nsubiron

asimay, 2017-12-14:
@coolKeen, how to reinstall graphical driver if no real nvidia graphic card?
I come across same issue. can you post your method at here?  
thanks!

asimay, 2017-12-14:
it required OpenGL 4.3, but I only got 3.3 version on my laptop.

nsubiron, 2017-12-14:
The OpenGL version depends on your GPU drivers, you can try to run the engine with the command-line argument `-opengl3`. However, if you are running CARLA in a laptop without GPU card, Unreal is unlikely to work. Check first if Unreal Editor is able to open

```
$ ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor -opengl3
```

coolKeen, 2017-12-14:
I have never installed gpu driver other than nvidia driver。You can search for suggestion on the Internet。

nina124, 2017-12-24:
hi @coolKeen , which nvidia driver did you install?
And after that, can you use opengl4.3 or still opengl3?

coolKeen, 2017-12-24:
1.
http://www.nvidia.cn/Download/index.aspx?lang=en
In this website,you can find the driver suitable for your gpu.
My gpu is GeForce GTX TITAN X and I just find the latest driver suitable for this in the website above.
2.
After that,I can use opengl4.5.
In Ubuntu,you can use the command "glxinfo | grep OpenGL" to see which version your gpu driver now support.

nina124, 2017-12-24:
@coolKeen Thank you so much!
After upgrading nvidia driver, opengl4.3 is successfully installed.

nsubiron, 2018-09-05:
Linking #757.

J-zin, 2019-03-30:
@coolKeen how did you fix it
did u update nvidia driver?
i got the same problem: ,it give me the information below which say my driver don't support opengl4.3. and my driver version is Driver Version: 384.81.
should i update the driver???(if so, i need to restart the server machine, but...it sounds impossible...) so, have any other method?

albertchristian92, 2019-11-04:
Hello, is there anyone has same problem with me:
I can update my driver into latest version: 440
But, my open GL showing 3.1. 
Anyone has met this issue before? Thanks.

LongruiDong, 2019-12-24:
> @nina124
> 1.
> http://www.nvidia.cn/Download/index.aspx?lang=en
> In this website,you can find the driver suitable for your gpu.
> My gpu is GeForce GTX TITAN X and I just find the latest driver suitable for this in the website above.
> 2.
> After that,I can use opengl4.5.
> In Ubuntu,you can use the command "glxinfo | grep OpenGL" to see which version your gpu driver now support.

It works for me!!
Thanks

pleasenott, 2024-06-06:
Hello, I run ./CarlaUE4.sh but just get this.
```
GameThread timed out waiting for RenderThread after 60.00 secs
Signal 11 caught.
Malloc Size=65538 LargeMemoryPoolOffset=65554 
CommonUnixCrashHandler: Signal=11
Malloc Size=65535 LargeMemoryPoolOffset=131119 
Malloc Size=139728 LargeMemoryPoolOffset=270864 
Engine crash handling finished; re-raising signal 11 for the default handler. Good bye.
Segmentation fault (core dumped)
```
and I find that your UE4 report so much?
which parameters can help me get more message .

Utine, 2024-07-15:
@pleasenott Did you solve this problem? I have same issue.