carla-simulator/carla #75: Missing Glass texture

nhynes, 2017-12-10:
When running CarlaServer 0.7.0, the log outputs a warning about missing the Glass texture and it seems like the streetlights are missing their glass casings. Is this the intended behavior?

Relevant log output:
```
LogMaterial: Warning: Material /Game/AutomotiveMaterials/Materials/Glass/M_Glass.M_Glass
missing bUsedWithInstancedStaticMeshes=True! Default Material will be used in game.
```

felipecode, 2017-12-12:
This seems like a bug ! 
Thanks a lot @nhynes .
Cheers.

nsubiron, 2017-12-12:
Yes, we are actually taking a look at this. They are two different issues, we already fixed the missing texture in the streetlights for the next release, and we are investigating why it gives warning in the glass material.

germanros1987, 2018-05-21:
How is this issue going?

nsubiron, 2018-07-12:
@TheNihilisticRobot any updates?

TheNihilisticRobot, 2018-07-25:
This issue was solved a while ago, i'm sorry.
I'll close this issue now.

closed 2018-07-25 by TheNihilisticRobot