carla-simulator/carla #78: Build Unreal Engine, original repo was gone!

jinfagang, 2017-12-12:
In this installation documentation: http://carla.readthedocs.io/en/latest/how_to_build_on_linux/ , the following line says:

```
$ git clone --depth=1 -b 4.17 https://github.com/EpicGames/UnrealEngine.git ~/UnrealEngine_4.17
```
Unforunetly the repo was gone, how to solve it.

yiakwy, 2017-12-12:
@jinfagang Unreal is a private repo. You need to sign up in Unreal official website and associate your Github account with Epic developer account.

closed 2017-12-12 by nsubiron

maxenceliu, 2017-12-25:
still can't dowload

`fatal: repository 'https://github.com/EpicGames/UnrealEngine.git/' not found`

ssokhey, 2018-08-24:
You have to accept the invitation from EpicGames from within your Github account.