carla-simulator/carla #79: The ./Util/download_from_gdrive.py does not work anymore

yiakwy, 2017-12-12:
1. This code was assumed to redirect user to download page and grant user a token for the link of the target file. However, it does not work anymore and connection is broken because of timeout.

2. I follow the discussion and try an alternative tool "gdrive". The tool also does not work after I submit the token.

```bash
(Py3) ➜  carla git:(master) ✗ gdrive download 11PKC_lG7L80L1ZxiMlzV17utU45UcRnT
Authentication needed
Go to the following url in your browser:
https://accounts.google.com/o/oauth2/auth?access_type=offline&client_id=367116221053-7n0vf5akeru7on6o2fjinrecpdoe99eg.apps.googleusercontent.com&redirect_uri=urn%3Aietf%3Awg%3Aoauth%3A2.0%3Aoob&response_type=code&scope=https%3A%2F%2Fwww.googleapis.com%2Fauth%2Fdrive&state=state

Enter verification code: 4/OBHu61Myoxr6WO1lIMza2iNy7Fw27TkiVAXvHNPLxmE
Failed getting oauth client: Failed to exchange auth code for token: Post https://accounts.google.com/o/oauth2/token: dial tcp 172.217.24.13:443: i/o timeout
(Py3) ➜  carla git:(master) ✗ vim ./Util/download_from_gdrive.py 
```

3. The code does not handle proxy properly

```bash
(Py3) ➜  carla git:(master) ✗ sh Setup.sh 
Backing up existing Content...
/Users/lei.wang1/Github/carla/Unreal/CarlaUE4/Content -> /Users/lei.wang1/Github/carla/Unreal/CarlaUE4/Content_20171212113735
downloading ...
file id is <11PKC_lG7L80L1ZxiMlzV17utU45UcRnT>
opening download page ...
Traceback (most recent call last):
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/connection.py", line 141, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/util/connection.py", line 83, in create_connection
    raise err
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/util/connection.py", line 73, in create_connection
    sock.connect(sa)
OSError: [Errno 65] No route to host
```

I have added the following codes to deal with it:

```
    print("file id is <%s>" % id)
    url = "https://docs.google.com/uc?export=download"

    session = requests.Session()

    print("opening download page ...")
    proxy = os.environ.get('SOCKS_PROXY', 'socks5://localhost:1080')
    proxies = {'http' : proxy,
               'https': proxy}
    response = session.get(url, proxies=proxies, params={'id': id}, stream=True)
```

To prepare the modification, one have to reinstall requests to support socks5 proxy:
> pip install 'requests[socks]'
> pip install --force-reinstall 'requests[security]'

4. bad handshake:
Something wrong with above handshakes
```bash

Traceback (most recent call last):
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 601, in urlopen
    chunked=chunked)
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 346, in _make_request
    self._validate_conn(conn)
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/connectionpool.py", line 850, in _validate_conn
    conn.connect()
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/connection.py", line 326, in connect
    ssl_context=context)
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/util/ssl_.py", line 329, in ssl_wrap_socket
    return context.wrap_socket(sock, server_hostname=server_hostname)
  File "/Users/lei.wang1/VirtualEnv/Py3/lib/python3.6/site-packages/urllib3/contrib/pyopenssl.py", line 448, in wrap_socket
    raise ssl.SSLError('bad handshake: %r' % e)
ssl.SSLError: ("bad handshake: SysCallError(-1, 'Unexpected EOF')",)
```

nsubiron, 2017-12-12:
The timeout error might be due to the terribly slow download speed that Google drive has in Asia (checkout #20). We were considering different options, like a mirror on Baidu, but we haven't implemented this yet in our scripts.

For now you can also try to download the content file directly from [Google drive](https://drive.google.com/open?id=11PKC_lG7L80L1ZxiMlzV17utU45UcRnT) from your browser using a proxy. Then extract it in "Unreal/CarlaUE4/Content".

Btw, we keep track of all the versions in the [ContentVersions.txt](https://github.com/carla-simulator/carla/blob/master/Util/ContentVersions.txt) file.

nsubiron, 2017-12-13:
Closing this, I created #84 to have a single place to discuss possible solutions to the download problem.

closed 2017-12-13 by nsubiron