carla-simulator/carla #80: Cannot open project in Windows with Unreal Engine 4.17.2

mikelortega, 2017-12-12:
Hi,
Which Unreal Engine version should I use?
I get these errors:
[image]
[image]
Thank you :)

nsubiron, 2017-12-12:
Hi @mikelortega,

Any Unreal 4.17 should work, for some reason since 4.17 some plugins appear as not compatible, but they are perfectly compatible. Just ignore the first message, press "No", you don't want to disable Carla.

The second message is just telling you that you need to compile the Carla code, press "Yes".

mikelortega, 2017-12-12:
Thank you Néstor, but following your instructions I get this error:
[image]

It seems it cannot compile the dlls:
LogInit: Warning: Incompatible or missing module: UE4Editor-CarlaUE4.dll
LogInit: Warning: Incompatible or missing module: UE4Editor-Carla.dll

[CarlaUE4-backup-2017.12.12-14.48.19.log](https://github.com/carla-simulator/carla/files/1551781/CarlaUE4-backup-2017.12.12-14.48.19.log)

nsubiron, 2017-12-12:
Can you try building with Visual Studio? Right-click in the "CarlaUE4.uproject" (in the folder "Unreal/CarlaUE4"), and select "Generate Visual Studio Project Files".  Then open the generated Visual Studio solution (*.sln) and build it. Hopefully the compiling error will give some hint on what fails.

mikelortega, 2017-12-12:
Ok, I see. The problem is I cannot compile CarlaServer, so carla/carla_server.h cannot be found.
When I try to compile CarlaServer there's something wrong with the protobuf folder, it doesn't contain the include folder that CarlaServer expects.

nsubiron, 2017-12-12:
Hmm yes, you need to compile several dependencies before you can open the project. Unfortunately we don't have official instructions to setup CARLA on Windows yet, it's totally possible but requires some extra work. Follow the discussion at #21 to do it.

However, if you have access to an Ubuntu machine I would recommend to build CARLA there as it's much easier, you can follow the instructions in the documentation ["How to build on Linux"](http://carla.readthedocs.io/en/latest/how_to_build_on_linux/).

nsubiron, 2017-12-13:
That file is generated by the protobuf compiler, you can generate it with the Makefile or calling directly "Util/Protoc.bat".

Lucaccio1, 2017-12-18:
@nsubiron I'm sorry to bother you right here but I'm having the same mikel's issue even if I'm building Carla in Linux. I'm confident I followed all the step correctly but unfortunately I had an error with "carla/carla_server.h" and with the creation of Lib4UEditor-Carla.sh

Can you help me figure out what's could be wrong?

nsubiron, 2017-12-19:
Hi @Lucaccio1, it seems like the setup went wrong. Try running Setup.sh and Rebuild.sh again, make sure you follow all the steps in ["How to build CARLA on Linux"](http://carla.readthedocs.io/en/latest/how_to_build_on_linux/).

Lucaccio1, 2017-12-19:
I rebuilt CARLA and server's issue diseappeared but CARLA's plugin still missing. That's the fatal error reported in the terminal. What's went wrong for your experience?

Thanks for replying

~/UnrealEngine_4.17/Engine ~/carla/Unreal/CarlaUE4
@progress push 5%
@progress pop
Performing 4 actions (8 in parallel)
[1/4] Compile Module.Carla.cpp
[2/4] Compile Module.Carla.gen.cpp
fatal error: file '/usr/include/linux/version.h' has been modified since the
      precompiled header
      '/home/clara/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/clara/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
fatal error: file '/usr/include/linux/version.h' has been modified since the
      precompiled header
      '/home/clara/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
      was built
note: please rebuild precompiled header
      '/home/clara/UnrealEngine_4.17/Engine/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/Engine/SharedPCH.Engine.h.gch'
1 error generated.
1 error generated.
ERROR: UBT ERROR: Failed to produce item: /home/clara/carla/Unreal/CarlaUE4/Plugins/Carla/Binaries/Linux/libUE4Editor-Carla.so
Total build time: 6,19 seconds (Local executor: 0,00 seconds)
LogInit: Warning: Still incompatible or missing module: libUE4Editor-CarlaUE4.so
LogInit: Warning: Still incompatible or missing module: libUE4Editor-Carla.so
LogExit: Preparing to exit.
LogModuleManager: Shutting down and abandoning module DesktopPlatform (36)
LogModuleManager: Shutting down and abandoning module AnimationModifiers (34)
LogModuleManager: Shutting down and abandoning module PropertyEditor (33)
LogModuleManager: Shutting down and abandoning module AudioEditor (30)
LogModuleManager: Shutting down and abandoning module TextureCompressor (28)
LogModuleManager: Shutting down and abandoning module ShaderCore (26)
LogModuleManager: Shutting down and abandoning module Landscape (24)
LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (22)
LogModuleManager: Shutting down and abandoning module OpenGLDrv (20)
LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (18)
LogModuleManager: Shutting down and abandoning module Renderer (16)
LogModuleManager: Shutting down and abandoning module Engine (14)
LogModuleManager: Shutting down and abandoning module CoreUObject (12)
LogModuleManager: Shutting down and abandoning module NetworkFile (10)
LogModuleManager: Shutting down and abandoning module CookedIterativeFile (8)
LogModuleManager: Shutting down and abandoning module StreamingFile (6)
LogModuleManager: Shutting down and abandoning module SandboxFile (4)
LogModuleManager: Shutting down and abandoning module PakFile (2)
LogExit: Exiting.
LogInit: Tearing down SDL.
Exiting abnormally (error code: 1)

nsubiron, 2017-12-19:
This happens from time to time due to Linux updates, you need to recompile the precompiled headers

    $ cd Unreal/CarlaUE4/
    $ make CarlaUE4Editor ARGS=-clean
    $ make CarlaUE4Editor

Lucaccio1, 2017-12-19:
First of all thank you @nsubiron cause I solved all my issues and now the Editor working. 

For sure I'm off the topic but may I ask you: CARLAUE4.sh it's not in my release package and I cannot find it neither at the end of the building process. Is it created during Setup.sh process or whenever?

nsubiron, 2017-12-19:
There is no "CarlaUE4.sh" script in the source version of CARLA, only in the release version. You run CARLA with the editor.

closed 2018-01-09 by nsubiron