carla-simulator/carla #82: Connection timed out after Setup.sh

jinfagang, 2017-12-13:
I just run ./Setup.sh and it always says connect time out:
```
➜  carla git:(master) ✗ ./Setup.sh                                
Backing up existing Content...
'/media/jintian/Netac/CodeSpace/ng/auto_car/system/carla/Unreal/CarlaUE4/Content' -> '/media/jintian/Netac/CodeSpace/ng/auto_car/system/carla/Unreal/CarlaUE4/Content_20171213111344'
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connection.py", line 141, in _new_conn
    (self.host, self.port), self.timeout, **extra_kw)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/util/connection.py", line 83, in create_connection
    raise err
  File "/usr/local/lib/python3.5/dist-packages/urllib3/util/connection.py", line 73, in create_connection
    sock.connect(sa)
TimeoutError: [Errno 110] Connection timed out

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connectionpool.py", line 601, in urlopen
    chunked=chunked)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connectionpool.py", line 346, in _make_request
    self._validate_conn(conn)
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connectionpool.py", line 850, in _validate_conn
    conn.connect()
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connection.py", line 284, in connect
    conn = self._new_conn()
  File "/usr/local/lib/python3.5/dist-packages/urllib3/connection.py", line 150, in _new_conn
    self, "Failed to establish a new connection: %s" % e)
urllib3.exceptions.NewConnectionError: <urllib3.connection.VerifiedHTTPSConnection object at 0x7f0c7e805eb8>: Failed to establish a new connection: [Errno 110] Connection timed out

```

What is this step wrong?

coolKeen, 2017-12-13:
If you are in China,I think this problem occur because of "wall".

jinfagang, 2017-12-13:
@coolKeen Well, Can I manully download from some link and import it into where it should to be.

coolKeen, 2017-12-13:
To be honest,I try downloading the content from  this link: https://zhuanlan.zhihu.com/p/30979943.But I failed to combine these content part.
So I just download the compiled project from https://github.com/carla-simulator/carla/releases/tag/0.7.0 and it runs well on my computer.For specific explanation,see #20 .

nsubiron, 2017-12-13:
Closing this, I created #84 to have a single place to discuss possible solutions to the download problem.

closed 2017-12-13 by nsubiron