carla-simulator/carla #84: Cannot download content from China (connection timeout in Setup.sh)

nsubiron, 2017-12-13:
The Setup.sh downloads the assets package (>10GB) from Google Drive. However, the download seems to be horribly slow from China (if not blocked), which results in a "connection timeout" error.

I am creating this issue to have a single place to discuss possible solutions.

- - -

The package can be directly downloaded from this link

  * [Content_0.7.0.tar.gz](https://drive.google.com/open?id=11PKC_lG7L80L1ZxiMlzV17utU45UcRnT)

Note that we keep track of all the versions in [ContentVersions.txt](https://github.com/carla-simulator/carla/blob/master/Util/ContentVersions.txt).

asimay, 2017-12-14:
content is too large to download... almost 12G, maybe.

jinlukang1, 2018-03-13:
@asimay 

Maybe a VPN(SSR) can help you。
You can try it.

YashBansod, 2018-03-13:
@nsubiron is it possible for your team to host the contents over multiple cloud platforms?
Maybe over onedrive, dropbox etc? How about hosting them over torrent?

Access to Gdrive was blocked in the organisation I work for and that caused me a lot of trouble while building carla from sources. One way around I found was to transfer it from Gdrive to onedrive for business (sharepoint) using multcloud and then download the contents from the sharepoint manually.

errorJY, 2018-04-28:
you can  use the IPV6 protocol to download it.

Liangsyl, 2018-05-02:
You can take a look at
https://zhuanlan.zhihu.com/p/30979943
There is a Content 0.6 which also works.

nsubiron, 2018-07-30:
This will hopefully be fixed with #573.

closed 2019-04-05 by nsubiron