carla-simulator/carla #85: No file of ./CarlaUE4.sh. and Need OpenGL 4.3, but that in ubuntu 16.04 is version 3.3.

asimay, 2017-12-14:
1. No file of ./CarlaUE4.sh. cannot find it.
2.  when `$ ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor "$PWD/CarlaUE4.uproject"`
Need OpenGL 4.3 support, but that in ubuntu 16.04 is version 3.3.
Find many materials by google to upgrade, even upgrade ubuntu kernel to 4.14. but all failed.
how to solve?

nsubiron, 2017-12-14:
1. The "CarlaUE4.sh" script is only present in the release version, see #22.
2. The OpenGL version depends on your GPU drivers, you can try to run the engine with the command-line argument `-opengl3`. However, if you are running CARLA in a laptop without GPU card, Unreal is unlikely to work. Check first if Unreal Editor is able to open

```
$ ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor -opengl3
```

closed 2017-12-14 by nsubiron

asimay, 2017-12-14:
@nsubiron ,thank you very much! but how to run carla simulator if build by source??
I can  open the ue4editor now.

asimay, 2017-12-14:
[image]

I download the **.tar.gz file, but no find the CarlaUE4.sh.

nsubiron, 2017-12-14:
Hi @asimay, you got the source tar.gz package instead; get the release (binaries) version [here](https://drive.google.com/open?id=1_r-UeNWAl6tHdtfRcYBrjfyVOtCz5Fbb). This one has the "CarlaUE4.sh".

In the source code there is no "CarlaUE4.sh", it is generated automatically by Unreal when the binary version is created. You need to open the "CarlaUE4.uproject" with Unreal Editor. See ["How to build on Linux"](http://carla.readthedocs.io/en/latest/how_to_build_on_linux/).

asimay, 2017-12-14:
dear @nsubiron , I build successfully, and all is OK, except last step. I also do according to what you instruct: add -opengl3 at last of comand, and it always core dump.

[image]

nsubiron, 2017-12-14:
Have you run "Setup.sh" and "Rebuild.sh"? 

If these two succeeded you can open the Carla project with
```
$ cd Unreal/CarlaUE4
$ ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor -opengl3 $PWD/CarlaUE4.uproject
```

asimay, 2017-12-14:
yes.  Setup.sh" and "Rebuild.sh is all work well. and successfully.

`$ ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor  "$PWD/CarlaUE4.uproject" -opengl3`

I used above command.

asimay, 2017-12-15:
hi, dear, @nsubiron  I tried your command:
`$ ~/UnrealEngine_4.17/Engine/Binaries/Linux/UE4Editor "$PWD/CarlaUE4.uproject" -opengl3`

everything load well, but at last it is core dump. what's the issue?
if the software has so uneasy solved issue occurs, I think it will not popularly accepted by anyone.
I tried almost 2 whole days for downloading and setup. but failed.

[image]

[image]

nsubiron, 2017-12-15:
Hi @asimay, I'm sorry to hear it still didn't work. Can you perhaps share your full log?

asimay, 2017-12-15:
[load_log2.txt](https://github.com/carla-simulator/carla/files/1562602/load_log2.txt)

yes. of course. thank you very much for your support.

nsubiron, 2017-12-15:
Uhm it's very puzzling, it doesn't give much info about why it crashes.

It is possible that your integrated GPU doesn't have enough memory/resources to handle all the content in Carla. Even if you manage to get it up and running, you are going to get very low performance. Unfortunately, Unreal and specially CARLA are very performance demanding.

nina124, 2017-12-24:
Hi, @asimay  is not the only one to meet this trouble.
I ran ./rebuild.sh but failed with opengl error，my computer is ubuntu 16.04 with opengl 3.3.
Then I tried -opengl3 as suggested, but meet core dump as well.
I tried to build for several days. Neither success to upgrade to opengl4.3, or build with -opengl3.
@nsubiron How do you upgrade ubuntu 16.04 with opengl4.3？
And the core dump problem should be treated.
PS, my graphic card is GeForce GTX 950.

nina124, 2017-12-24:
Problem solved with coolKeen's instruction in #73

asimay, 2018-01-08:
I have added the memory to 16G-24G, but it seems still crash... the error like above..
this is not a good  and suitable tool for general using. I think. Maybe I will give up.

nina124, 2018-01-08:
Hi @asimay 
You could try the release version first:
1. download [here](https://drive.google.com/file/d/1_r-UeNWAl6tHdtfRcYBrjfyVOtCz5Fbb/view) and extract it
2. if you need to run CARLA off-screen(e.g. a ubuntu server, log in with `ssh`), install the [VirtualGL](https://sourceforge.net/projects/virtualgl/files/2.5.2/) and [TurboVNC](https://sourceforge.net/projects/turbovnc/files/2.1.1/)
3. try 
`./CarlaUE4.sh /Game/Maps/Town01 -carla-server -benchmark -fps=15 -windowed -ResX=800 -ResY=600`
`./client_example.py -a`
If it successes, the CARLA server is running, and the python client use the built-in autopilot to drive the car.

If you still need to build from source, follow coolKeen's instruction in #73:
1.  http://www.nvidia.cn/Download/index.aspx?lang=en
In this website,you can find the latest driver suitable for your GPU.
2. My GPU is GeForce GTX 1080. After step 1, my GPU driver supports opengl4.5.
You can use the command "glxinfo | grep OpenGL" to see which version your gpu driver now support.
3. After gpu dirvers supports opengl4.5, then build from source following the documentation again. It should success now.

I did not manage to build with `-opengl3`. I think it is unstable, meeting segmentation fault as well. And I do not think the GPU memory is the cause. The GeForce GTX 1080 only has 8GB memory.

asimay, 2018-01-08:
@nina124 ,hi, dear,  I 've already installed successfully. only when load UE4 editor the program crash. as @nsubiron said, the issue is caused by memory limited, but I added up to 16-24G of cpu memory, it is still crash when load ue4.

the error is the same with this:
[image]

nsubiron, 2018-01-08:
Hi @asimay, I meant not enough GPU resources. I think you are using the integrated graphics of your laptop, are you? Unreal Engine usually doesn't work on such set up, it is highly recommended to have a dedicated graphics card for it, see [Recommended hardware](https://wiki.unrealengine.com/Recommended_Hardware). It's unfortunate, but at this point you need a very powerful computer to run CARLA.

asimay, 2018-01-08:
@nsubiron ,ok, I assumed it was cpu memory.

actually I have this graphic card, one is integrated, one is AMD R5.

[image]

nsubiron, 2018-01-08:
@asimay There are many things that can go wrong, I'm sorry for not being of much help. The AMD Radeon R5 M430 is a low end GPU and might still not work, make sure your card is selected as default graphics unit and you have the right NVIDIA drivers installed for that card. As a last resort, you can debug it with GDB, that might give some hints on what fails.

nsubiron, 2018-09-05:
Linking #757.