carla-simulator/carla #87: Adding point to spline in SplineMeshRepeater

curantil, 2017-12-14:
I'm trying to use the SplineMeshRepeater, but I cannot add points to the spline. 

I tried right clicking and selecting "Add point" when a segment was selected and also selecting an existing point and duplicating it, but in both cases, it seems to create a point that is not connected to the spline or do nothing at all. The disconnect point also disappears when selecting anything else other than the point.

nsubiron, 2017-12-14:
Hi @curantil,
You can drag and drop spline points while holding "alt" key

[image]

curantil, 2017-12-14:
Thanks! That works indeed.

closed 2017-12-14 by nsubiron