carla-simulator/carla #88: Steering wheelbrake joystick with Carla

sandman, 2017-12-14:
Hi,
Has anyone successfully managed to set up (and calibrate) a steering wheel and brake joystick (like the Logitech Driving Force GT for example) with Carla? 

I'm trying to set one up right now and would like to learn of any pitfalls.

sandman, 2017-12-14:
So I got it to work: https://gist.github.com/sandman/366e45d3a836da1a2a90fe9eeccd689a

pygame makes this super easy.

sandman, 2017-12-14:
Sorry for the 'hacky' code. I can refine it over time. But essentially the joystick input depends on the model and so it could be difficult to make a generalized implementation. But using [jstest ](https://github.com/Grumbel/jstest-gtk) to query the joystick model, a proper commit can be made.

felipecode, 2017-12-15:
Hey @sandman , thanks for the interest.
I did this myself also ! For the Logitech G29, using pygame, in a similar way than you.
But the code is even more hacky than yours, so we didn't release.
The problem is, as you said, each joystick is different, so you need to
re-adapt the keys. And people would want to configure the mapping from joystickinput
to steering in a way they like like.
A jstest version would probably be better. 
We could eventually think of something running over the client to provide support for driving with joysticks in a nice way, joining what you said on issue #89

sandman, 2017-12-15:
Hey @felipecode , nice that you got it to work! How about simply adding a short HOWTO: in the Documentation and people who are interested can get working on it? I can help with this.

Another issue I am facing now is the calibration of the steering but mainly the throttle and brake pedals. I think it is not so easy! Any ideas how to do it scientifically?

zhangjun-xyz, 2018-03-05:
@sandman  Which steering wheel did you use? Logitech G920 or Logitech G29?
Can you offer more details?
Thanks!

sandman, 2018-03-05:
@zhangjun-xyz I'm using the Logitech Driving Force GT. But another model should work just as well. You can check out my gist posted earlier for the code.

zhangjun-xyz, 2018-03-06:
@sandman Thanks for your kind reply. :+1:

fbuchinger, 2018-04-14:
Hi @sandman, I would also be interested in using my Logitech G25 steering wheel with Carla. Unfortunately  your gist doesn't seem to work anymore with Carla 0.81 and Python 2.7.11:

```  File "manual_control_steering_wheel.py", line 42, in <module>
    from carla import CARLA
ImportError: cannot import name CARLA```

To me it seems the Python Client API has changed or is this a Python 2.7 vs 3.3 thing?

sandman, 2018-04-16:
Here's an updated gist that works in 0.8.1: https://gist.github.com/sandman/24970f0a6fe4d40293ff86159ee740cb

Note that only the throttle and steering commands are used from the steering wheel. For additional commands (like reverse, scene change etc.) you would need to map the joystick input to the corresponding control. This can be done in a straightforward way with jstest.

fbuchinger, 2018-04-17:
Thank you! Will check it out tonight.

sandman, 2018-04-17:
By the way, currently its very difficult to steer the car comfortably as the wheel is not properly calibrated and there is no official kernel support for Force Feedback. I have a naive linear steering control that does not perform well in practice. If you have a solution for this, please share with us!

fbuchinger, 2018-04-17:
@sandman seems that there are undefined globals in lines 246 to 254:

```
>python manual_control_steering_wheel.py
STARTING in a moment...

Starting new episode...
Joystick ID: 0 Init status: 1 Axis(1): 0
Steering angle: 0.000000 Throttle: 0.000000 Brake: 0.000000
Traceback (most recent call last):
  File "manual_control_steering_wheel.py", line 461, in <module>
    main()
  File "manual_control_steering_wheel.py", line 450, in main
    game.execute()
  File "manual_control_steering_wheel.py", line 163, in execute
    self._on_loop()
  File "manual_control_steering_wheel.py", line 246, in _on_loop
    print('reverse: %d toggle: %d' % (reverse, toggle))
NameError: global name 'reverse' is not defined
```
Script ran after I commented out the mentioned lines. Nevertheless I couldn't start the car properly due to the low frame rate (~7 fps).

I was also wondering whether https://github.com/schmerdy/openxc-wheel could be useful for the steering wheel support in carla.

fbuchinger, 2018-04-18:
Regarding better steering wheel support, I'd recommend to check out the logitech-g27 library for nodejs. 
It offers a more advanced steering wheel control routine that differentiates between coarse and fine movements: https://github.com/ForestMist/logitech-g27/blob/master/code/data-map.js#L254
The library also supports force feedback control (see scripts in test folder).

felipecode, 2018-05-01:
Hey guys any progress on that ? We would be interested into
incorporating an interface for human driving as a PR. 
CHeers !

fbuchinger, 2018-05-01:
Hi @felipecode! I think @sandman's script is a step in the right direction, but not yet ready for a PR. I had to comment out some lines to get it to run (see above), as a consequence it doesn't have reverse gear logic.

But the bigger issue is the lack of compatibility between the different (Logitech) steering wheels. My Logitech G25 wheel has a completely different pedal/button layout than @sandman's Logitech GT - as a consequence, I could only use the steering wheel itself to control the car, but not the pedals, gear shift, button etc.

[OpenXC's wheel.py class](https://github.com/schmerdy/openxc-wheel/blob/master/wheel.py) clearly shows that PyGame's Joystick control is too low-level/primitive for true steering wheel control. Unfortunately this class only works for G27 wheels and is unmaintained. I'm afraid you would have to invest quite some work to get it to run with other Logitech Wheels.

sandman, 2018-05-03:
Hi @fbuchinger, I agree that the script I provided is just a quick hack to map the essential controls for driving. Note that not all driving controls may be necessary in CARLA. For example: Assuming automatic transmission, the necessary  accelerator and reverse gear are needed together with steering and brake. Hence there is no need to comprehensively map *all* controls from all wheels IMO. The focus can be to support a few wheels completely rather than all wheels completely. What do you think @felipecode ?

Coming to the real issue which is steering wheel calibration. CARLA takes input values from -1.0 to 1.0 unlike games like ETS/GTA etc. which have a higher range. Same applies for the throttle and brake. So the effective range is limited, making the wheel very sensitive. It would be great if we could come up with a smart solution for this..

fbuchinger, 2018-05-06:
Hi @sandman, I agree with you that we should only port the necessary controls wheel, throttle, brake and reverse. Unfortunately, the pedal layout differs quite a lot between the different wheels.

The most simple solution i can think of would be a `wheel.ini` file with sections for every wheel that maps the input to every command (inspired by openxc's [wheel_config.py](https://github.com/schmerdy/openxc-wheel/blob/master/wheel_config.py)):
```ini
[G27 Racing Wheel]
steering_wheel = 0
accelerator = 1
brake = 2
clutch = 3
gear_1 = 12
gear_reverse = 22

[G25 Racing Wheel]
#config for Logitech G25
```
(the section title corresponds to [pygame's joystick](https://www.pygame.org/docs/ref/joystick.html#pygame.joystick.Joystick.get_name) name)

In order to create mappings for unknown wheels, we could add a cli option `-m` or `--mapwheel`. It would interactively ask the user to perform the required actions:
```
> python manual_control.py --mapwheel
Unknown wheel 'Logitech G25' detected.
Please press accelerator pedal.
Axis 2 configured for accelerator.
Please press brake pedal.
....
```
The mapwheel command would simply append the configuration to the ini file.

What do you think?

felipecode, 2018-05-09:
Hello Guys,
I like the @fbuchinger, solution for having different wheels.
It think mapping steering , throttle, brake reverse gear and hand brake is enough.

About the wheel sensitivity, i didn't see any problems when I was working with
it. You just need a good mapping function from the wheel output signal to -1,1. 
Some soft exponential function with dead zones maybe ?  

To create the autocentering and force feedback i was using this library
https://github.com/MadCatX/FFBChecker. Maybe we could find a cleaner solution ?
We will have some people working on this in fulltime in house for a while. 
@rohitgajawada and @madan96 
Do you already have a fork where we can contribute ? We will make one
if not. 
Thanks a lot guys !

fbuchinger, 2018-05-11:
Hi @felipecode,

thanks for the feedback! 

About autocentering/force feedback: the [node.js driver for the Logitech G27](https://github.com/ForestMist/logitech-g27) supports setting [these](https://github.com/ForestMist/logitech-g27/blob/master/docs/api.md#force), it is directly built upon [node-hid](https://github.com/node-hid/node-hid), which provides low-level access to USB-based human input devices.

Maybe you could use [PyUSB](https://github.com/pyusb/pyusb) for a similar approach (which would be pretty low-level though and requires testing of each wheel). 

AFAIK there is no steering wheel for for CARLA, @sandman just created a gist containing his modified script.

felipecode, 2018-05-16:
Some folks released a first version for joystick control for logitech G29.  It was very based on @sandman  GIST.
https://github.com/rohitgajawada/carla/tree/intern/PythonClient
 The example can be run on "human_controller.py" . It uses the ini files as proposed by @fbuchinger .
On this fork, I can see the following issues:

1. Manual control and human control now are duplicate code. We will try to join both. 
2. We need an integrated solution to have the autocentering and the force feedback without external dependencies. Thanks @fbuchinger  for the suggestions. 
3. Pygame is very slow #421.
4. Having a HUD as described in  issue #89 would be cool. Has anyone worked on that ?

Cheers

fbuchinger, 2018-05-17:
@felipecode thanks to you and your team for implementing the prototype so quickly! Will check it out tonight.

Regarding (3): I think it's important to find out why PyGame is slow in our scenario. Is the slowness caused by the CPU/GPU? Then it could make sense to use another 3D engine within Pygame (ModernGL comes to my mind). If the poor performance is caused by IO (e.g. blocking socket.read() calls that delay image downloads), then we should switch to an async networking framework (e.g. Twisted or Python 3 asyncio).

fbuchinger, 2018-05-18:
Another thought on the performance issue: have you tried to run the client script on pypy instead of cpython? Pypy 6 was just released and offers full pygame compatibility along with up to 30x speedup: http://renesd.blogspot.co.at/2018/03/pygame-on-pypy-usable.html?m=1 (in some cases it might also be slower though)

fbuchinger, 2018-05-20:
I did a first check on humancontrol.py and added the Logitech G25 to the wheel config: https://github.com/fbuchinger/carla/commit/40ad9ab5b3723f4200da570566b0ded8f44e2f4a
Unfortunately, the code runs into an exception when connecting to carla server - any hints on that?

madan96, 2018-05-20:
Could you please post the exception you're getting? It'd be easier to debug that way.

fbuchinger, 2018-05-21:
@madan96 here is my python 2.7 traceback when running https://github.com/fbuchinger/carla/blob/intern/PythonClient/human_control.py
```

> python human_control.py
INFO: listening to server localhost:2000

Welcome to CARLA manual control.

Use ARROWS or WASD keys for control.

    W            : throttle
    S            : brake
    AD           : steer
    Q            : toggle reverse
    Space        : hand-brake
    P            : toggle autopilot

    R            : restart level

STARTING in a moment...

Starting new episode...
Traceback (most recent call last):
  File "human_control.py", line 486, in <module>
    main()
  File "human_control.py", line 478, in main
    game.execute()
  File "human_control.py", line 235, in execute
    self._on_loop()
  File "human_control.py", line 274, in _on_loop
    measurements, sensor_data = self.client.read_data()
  File "c:\Users\fbu\dev\carla\PythonClient\carla\client.py", line 127, in read_data
    return pb_message, dict(x for x in self._read_sensor_data())
  File "c:\Users\fbu\dev\carla\PythonClient\carla\client.py", line 127, in <genexpr>
    return pb_message, dict(x for x in self._read_sensor_data())
  File "c:\Users\fbu\dev\carla\PythonClient\carla\client.py", line 175, in _read_sensor_data
    yield self._parse_sensor_data(data)
  File "c:\Users\fbu\dev\carla\PythonClient\carla\client.py", line 180, in _parse_sensor_data
    return parser.name, parser.parse_raw_data(data[4:])
  File "c:\Users\fbu\dev\carla\PythonClient\carla\client.py", line 196, in parse_image
    return sensor.Image(frame_number, width, height, image_type, fov, data[24:])
  File "c:\Users\fbu\dev\carla\PythonClient\carla\sensor.py", line 149, in __init__
    assert len(raw_data) == 4 * width * height
AssertionError

```

fbuchinger, 2018-05-23:
@madan96 any updates on my error? Will try to look into it again next weekend. BTW I noticed your "intern" branch (which i forked) has disappeared. Should I wait for your merge to develop and then re-create a fork based on develop?

madan96, 2018-05-28:
@fbuchinger Sorry for the delay. I was trying to reproduce the error but couldn't do it. My client side is updated with the master and I'm using the binaries available on GitHub for running the server side. Since the error occurs while receiving sensor data, there's a bit of confusion that it might be due to the server side. Which binary are you using for CARLA server? Did you do a build from master branch?

fbuchinger, 2018-05-29:
@madan96 my bad, I was still using the 0.8.1 release. Now the error has gone. Which branch are you using now for the joystick support?

madan96, 2018-05-30:
@fbuchinger For now the code is still in https://github.com/rohitgajawada/carla/tree/intern/PythonClient. But we are working on improving the code and we'll soon try to have it on https://github.com/carla-simulator/carla/tree/CS1_client_side_agents

fbuchinger, 2018-06-10:
Just saw that 0.83 was released on Friday. Will test my code with that version and submit a PR against the intern branch then.

imran514, 2018-07-17:
I am trying to use Logitech G920, can any give me some input  where should start from?
Does @fbuchinger code still works ? can i use it for G920 too ?

fbuchinger, 2018-07-17:
Hi @imran514! At the moment you would have to download [human_control.py](https://github.com/rohitgajawada/carla/blob/intern/PythonClient/human_control.py) and [wheel.ini](https://github.com/rohitgajawada/carla/blob/intern/PythonClient/wheel_config.ini) from Rohits CARLA fork. Then you start CARLA and the human_control.py client. 

Now steer the wheel and push the throttle/brake pedals. Grep for the `Joystick ID:` log output and take note of the axis ID for the wheel and each pedal. Then open wheel_config.ini and adjust them.

Next, push the button on your steering wheel that should act as a handbrake and take note of its jsButton ID. Then switch to the reverse gear and do the same (Hint: watch out for the log line after `Steer Cmd` to obtain the button ids). Then add both button ids to wheel_config.ini and you should be good to go.

rohanpansare, 2018-07-25:
@fbuchinger and Rohit, Thanks for the scripts. I am able to map it to Fanatec and get it working as intended. But, I did observe an odd behavior where the vehicle loses focus and does not respond to any controls after a few seconds. And that behavior was also common when I ran the manual_control.py script for WASD controls. I do see the logs and their event values getting changed 
 as you mentioned above but the car doesn't respond to any input after a while. Do you know why that may be happening?

germanros1987, 2018-09-20:
Hey...how about finishing a PR of this...? :P

sangeetsu, 2019-02-13:
@imran514 Did you get the Logitech G920 to work? If so please explain what you did.

germanros1987, 2019-02-13:
checkout the branch features/steering_wheel_control
<https://github.com/carla-simulator/carla/compare/features/steering_wheel_control>
.
I just added code to use the steering wheel in manual mode.

sangeetsu, 2019-02-16:
@germanros1987 With a little help from this http://www.draisberghof.de/usb_modeswitch/bb/viewtopic.php?f=2&t=2510 and your code, I got the G920 working on Carla. Thank you!

sangeetsu, 2019-03-16:
After updating to 0.9.4, I'm getting the following error 
`Traceback (most recent call last):
  File "manual_control_steeringwheel.py", line 855, in <module>
    main()
  File "manual_control_steeringwheel.py", line 847, in main
    game_loop(args)
  File "manual_control_steeringwheel.py", line 784, in game_loop
    world.tick(clock)
  File "manual_control_steeringwheel.py", line 176, in tick
    self.hud.tick(self, clock)
  File "manual_control_steeringwheel.py", line 421, in tick
    'Map:     % 20s' % world.world.map_name,
AttributeError: 'World' object has no attribute 'map_name'`
This happens when I run manual_control_steeringwheel.py
Can someone explain the error? Thank you.

mallela, 2019-03-27:
@felipecode I'm having trouble understanding the parameters to be set and the effect it will have on force feedback/autocentering in FFBChecker. Is there any helpful link you know of that discusses the parameter description? I'm using the Logitech G29. I experimented with some autocentering and gain values but don't understand the other parameters (replay length/delay, repeat, attack level/length fade level/length).

damascenodiego, 2019-05-03:
Dear members,

I managed to set values to the FF_AUTOCENTER parameter of the Logitech steering wheel using the [python-evdev](https://github.com/gvalkov/python-evdev) library. 

Here it follows [the code](https://gist.github.com/damascenodiego/7fcac91e44033a655f8c89ba2aebfd9a) to test the autocenter feature.

Now, my question is: How can we associate this parameter to make the driving experience a little more realistic?

walzimmer, 2019-07-05:
@damascenodiego Thank you for the code snippet. It will set force feedback to the maximum if setting val = 65535. We need to add this code into the game loop of manual_steering_wheel.py and make it dependent on the steering wheel and the velocity.
E.g. 
device = evdev.list_devices()[0]
evtdev = InputDevice(device)
while True:
  if steering_angle > 0.75 and velocity > 100:
    val = 65535
  elif steering_angle > 0.5 and velocity > 50:
    val = 65535 * 0.5
  else:
    val = 0
   evtdev.write(ecodes.EV_FF, ecodes.FF_AUTOCENTER, val)

damascenodiego, 2019-07-05:
That's exactly my idea. :)

I'll reverse my code which uses updated libraries updated to the versions
that Carla is using now and I'll do my push request.

soulslicer, 2019-10-15:
Hey is there a version that works with the newest version of CARLA?

soulslicer, 2019-10-21:
@damascenodiego 

Is there now a version out with force feedback? I have a logitech g29 and there is no force feedback with the manual_control_steeringwheel.py

damascenodiego, 2020-01-09:
Hi everyone, I have implemented some primary implementation of force feedback using a previous version of the Carla script manual_control_steeringwheel.py.

This can be found in my [repo's fork](https://github.com/damascenodiego/scenario_runner-v095/blob/scenario_ui_merged/manual_control_steeringwheel.py).

soulslicer, 2020-01-09:
Does it behave like in real life? Like if I am turning and halfway I stop, the steering should stop turning. And if i continue to press on the acceleration, the wheel should turn automatically

damascenodiego, 2020-01-09:
No, unfortunately, it doesn't.
This is a very primitive version that just centers the wheel accordingly to the speed.
The faster you are, the higher is the value set to FF_AUTOCENTER.

aakashsehgal, 2020-01-15:
@damascenodiego I tried this with carla 0.9.6 (Windows) and it works...a small correction : line 527 should be `world.world.get_map().name`
The steering works but somehow the pedals dont seem to work. I have rechecked the connections but didnt help. Any ideas ?

damascenodiego, 2020-01-20:
Unfortunately, I didn't try this code on Windows.
The only issues I had with pedals happened due to the switch button.
Depending on its position, the mapping between commands and buttons/pedals
will change and hence, it may not work properly.

damascenodiego, 2020-01-20:
Unfortunately, I didn't try this code on Windows.
The only issues I had with pedals happened due to the switch button.
Depending on its position, the mapping between commands and buttons/pedals
will change and hence, it may not work properly.

On Wed, Jan 15, 2020 at 3:39 PM aakashsehgal <notifications@github.com>
wrote:

> @damascenodiego <https://github.com/damascenodiego> I tried this with
> carla 0.9.6 (Windows) and it works...a small correction : line 527 should
> be world.world.get_map().name
> The steering works but somehow the pedals dont seem to work. I have
> rechecked the connections but didnt help. Any ideas ?
-- 
Diego Damasceno

ym2509, 2020-11-17:
Hello!
I'm running the Complied/Shipping CARLA v0.9.9 on linux and trying to use a Logitech G27 steering wheel to control the car. I have set the buttons using jstest-gtk and have a wheel_config.ini file that maps the buttons and axes but the axes/button mapping does not work when I run the client. 
Would greatly appreciate any help!
Thank you

erfpak7, 2021-03-02:
> After updating to 0.9.4, I'm getting the following error
> `Traceback (most recent call last): File "manual_control_steeringwheel.py", line 855, in <module> main() File "manual_control_steeringwheel.py", line 847, in main game_loop(args) File "manual_control_steeringwheel.py", line 784, in game_loop world.tick(clock) File "manual_control_steeringwheel.py", line 176, in tick self.hud.tick(self, clock) File "manual_control_steeringwheel.py", line 421, in tick 'Map: % 20s' % world.world.map_name, AttributeError: 'World' object has no attribute 'map_name'`
> This happens when I run **manual_control_steeringwheel.py**
> Can someone explain the error? Thank you.
> **EDIT**(fix) - map_name was removed from world, use world.get_map().name (c/o @nsubiron)

I'm getting the same error trying to use Logitech G29 on Windows 10. Is there any solution for it?

damascenodiego, 2021-03-02:
Unfortunately I don't have access to a steering wheel anymore. Then, it's
difficult to help you out.

However, given the last message, I would say that you could try updating
the *map_name* attribute to the corresponding one in the version you are
using, or simply commenting this part of the code.

yshen47, 2021-03-26:
@erfpak7 Easy fix, just comment that line out

erfpak7, 2021-03-26:
> @erfpak7 Easy fix, just comment that line out

@yshen47 Thanks! I already figured but I thank your consideration. 
Do you have any thought on this issue though : https://github.com/carla-simulator/carla/issues/3976

VincentCheung991214, 2021-11-23:
Hi everyone,
I wanna modify the code such that achieving below purposes:
1. Delay the frame for .1-1 seconds, like the hero car already turned right at intersection, but the frame of driving terminal is delayed for 0.5 seconds to show its right turning. 
2. also delay the time for sending driving wheel control's command: like I step on the break, it takes 0.5 seconds more to send the stop command to the server.

I tried many methods but all I can received is a deduced FPS. However, I need to delay the frame and the time for signaling the car control system, instead of having a laggy screen (low FPS).

For the 1. target, I did try making some changes in _self.frame = timestamp.frame_ , but nothing was happened.
For the 2. , I intentionally add a time.sleep in the _parse_events_ but it is not workable and seems making ssuspension to other running blocks.

Plz give me some directions if getting any clues on it. Thanks and appreciate.

niezifei, 2022-03-19:
Hi, everyone, I was trying to connect the Fanatec SteeringWheel with Carla on Windows 10, an error reports that  "Please Connect Just One Joystick", when I run the manual_control_steeringwheel.py? 
Anyone knows how to fix it?

hisashiim, 2022-08-05:
@niezifei 
Hi
I had the same problem.
I checked the following site and found multiple indices detected.
https://gamepadtest.com/

        joystick_count = pygame.joystick.get_count()
        if joystick_count > 3:
            raise ValueError("Please Connect Just One Joystick")

        self._joystick = pygame.joystick.Joystick(1)

I have modified some of the .py.
It worked successfully.

niezifei, 2022-08-08:
@hisashiim 
Thank you for your info.
As you said, I tried to modify the code in `manual_steeringwheel.py`. But still, it cannot identify the steering wheel and pedals of Fanatec.
Did you use the Fanatec steering wheel as well? Could you please share your code and wheel_config.ini ? 
Thank you.

hisashiim, 2022-08-08:
@niezifei 
First, let's check each other's development environments.
My System Environments
-Windows10
-Carla 0.9.10
-Fanatec https://simracingsetup.com/shop/bundles/xbox-one-competition-pack/
The settings of wheel_config.ini differnt depending on the device.

With the equipment connected, try accessing the following site.
Is the device recognized?
Can you give us a screenshot?
https://gamepadtest.com/

[2 more replies]