carla-simulator/carla #89: Showing Tachometer and Speedometer in driving view with HUD

sandman, 2017-12-14:
Hi,
It would be very useful to show the speedometer and tachometer like a real car in the driving window. I guess this should not be too difficult as the info is already available in the HUD. Speaking of which, is there a way to get the HUD overlaid on the camera view and not in a separate window?

Thanks,
Sandip

felipecode, 2017-12-15:
I was doing something similar... But no speedometer or tackometer.
I was using pygame. It is easy to use but super slow !
I will try to organize and share here the code next week
You can see on this video exactly what I have :
https://www.youtube.com/watch?v=cFtnflNe5fM

sandman, 2017-12-15:
Very cool video! I was thinking something on the lines of a realistic driving cockpit like so [https://s4-frame.ozstatic.by/1000/236/108/20/20108236_2.jpg](url)

Maybe the wheel movements could be filtered over multiple frames to reduce render processing time? Awaiting your code in any case!

nsubiron, 2017-12-15:
We could extract the engine rotation speed from the [UWheeledVehicleMovementComponent][complink] and easily display it in the HUD as number (a nicer view would be nice but it's not really a priority right now). We can consider sending it to the client too. However, I don't know how realistic are the simulated RPM and if it would be of any practical use.

[complink]: https://docs.unrealengine.com/latest/INT/API/Plugins/PhysXVehicles/UWheeledVehicleMovementComponent/

sandman, 2017-12-15:
Thanks, nsubiron ! I think the main benefit is to know the current speed of the car, in order to adjust the control to obey the traffic signs. The problem is that the HUD is displayed in another window (server?) but the driving window is different. So it kind of defeats the purpose of the "HUD". Unless I am missing something here?

[image]

nsubiron, 2017-12-15:
Yes, the other window is the server window. That HUD is mostly to cross-check the info received by the client, also useful for testing in non-networking mode.

The manual control script is a sample of what you can see in the client side using pygame, the idea is that people will adapt this script to their needs. We would be happy to incorporate your changes if you find a nice display for it :)

danielcomas25, 2018-08-14:
@nsubiron @felipecode @dehabu Hello, I have been working with Carla for a few weeks and I find it very good. However, I wanted to implement a longitudinal controller taking dynamic variables such as torque in the wheels, angular velocity and RPM. 

In the Carla paper published in your webpage [carla.org](http://proceedings.mlr.press/v78/dosovitskiy17a/dosovitskiy17a.pdf) , a modular pipeline was implemented. In this pipeline (section 3.1), there is a description about the Continuous Controller and there is a description about the PID controller that you implemented, in this section there is a reference to [another paper](https://downloads.hindawi.com/archive/2014/259465.pdf) in which you used a Roboust PID controller. According to this controller you modeled your Internal Combustion Engine using a static engine map that defines the relationship between the input throttle position and the engine speed to calculate the output torque (section 2.3 equation 8).

That is pretty much what I am trying to do here in order to implement the logitudinal controller. So, I need to find the relation between those variables. However, I need a way to acces them, specially the RPM value in order to know the torque. Can you give me a hint on how can I acces RPM, angular velocity or toque applied to the wheels and send them to the python client? I am looking forward to implement a controller but I need this variables in order to have a correct dynamic model of the vehicle to apply my controller.

I would like to thank you for your answer and for the hard work done so far..!

nsubiron, 2019-04-05:
Closing this, we have a HUD with a lot of info already.

closed 2019-04-05 by nsubiron