carla-simulator/carla #90: Mixed reality overlay

sandman, 2017-12-15:
Hi,
Would it be possible to overlay the semantic segmented map on top of the Post-processed camera view to create a mixed-reality view?

nsubiron, 2017-12-15:
I think this should be done in the client side, it shouldn't be too difficult to blend the two images in Python

felipecode, 2017-12-15:
I don't exactly understand your objective.  What do you mean by mixed-reality view ?
Just a simple merge between labels and rgb ? This could be easy to be done on the client
side. If you do it, maybe it is a good function to eventually add to the client.

sandman, 2017-12-15:
I'm exploring shared semi-autonomous control where a human driver is assisted by local-AI in the car. Here's a reference: http://ieeexplore.ieee.org/abstract/document/6232153/

In this case, a simple merge with a transparency layer should work, right?

felipecode, 2017-12-15:
Yes ! It should work! 
Please share with us after you do it !

sandman, 2018-01-24:
Sorry, I mistakenly tried to commit to the master! I got this working with a simple 1-line change. You can view the gist --> https://gist.github.com/sandman/9549ebc5188e493cc4091fe268accdb5

sandman, 2018-01-24:
And here's a screenshot:
[image]

nsubiron, 2018-01-25:
Hi @sandman, thanks for sharing! :)

closed 2018-02-01 by carla-admin