carla-simulator/carla #91: Is there a way to control a pedestrian?

DanielLSM, 2017-12-16:
I was looking into the documentation, and could not see an answer to this question.
How one should proceed to control a pedestrian?

nsubiron, 2017-12-19:
This is not yet possible. At spawn, pedestrians are given a target location in the world (by AWalkerSpawnerBase) and they find their way with Unreal's path-finding system.

closed 2017-12-21 by DanielLSM

hs105, 2018-03-05:
Could you clarify the "target location" is actually the spawning point where the pedestrian will start?
 
"At spawn, pedestrians are given a target location in the world (by AWalkerSpawnerBase)"

nsubiron, 2018-03-05:
@hs105 The walker spawner choses at random two points from the list of _"Walker Spawn Points"_ present in the map, say `A` and `B`. Then spawns the walker at `A` and calls `MoveToLocation(B)` function from [AAIController](https://docs.unrealengine.com/latest/INT/API/Runtime/AIModule/AAIController/). When a walker reaches `B` it gets destroyed. Thus every walker's path starts and ends at a _"Walker Spawn Point"_.

_"Walker Spawn Points"_ are strategically located behind buildings so you won't normally see pedestrians appearing and disappearing (though you can see this sometimes when a pedestrian gets stuck, but that's a known issue we hope to fix at some point). In the map there are also _"Walker Start Spawn Points"_, these are only used at begin play to quickly populate the city.

hs105, 2018-03-05:
Great! thanks for your quick reply!!

How do we set "Walker Spawn Points"? Do we need to access the map? We're new to carla. We appreciate if you could give a pointer on this. 

Also, in the python code "client_example.py", what the "NumberOfPedestrians" controls?

nsubiron, 2018-03-05:
"Walker Spawn Points" can only be edited from within Unreal Editor. You'd need to compile CARLA from source and open the CarlaUE4.uproject, then drag-and-drop the spawn points in the map.

> what the "NumberOfPedestrians" controls?

Controls the total number of walkers that the walker spawner is going to try to have spawned at any time. Note that at times the number of spawned walkers can be lower, for instance if all the spawn points are momentarily blocked by other walkers it won't be able to spawn a new one.

All the settings in the Python examples correspond to the settings in the ["CarlaSettings.ini"](https://github.com/carla-simulator/carla/blob/master/Docs/Example.CarlaSettings.ini)

```ini
; Number of non-player pedestrians to be spawned into the level.
NumberOfPedestrians=30
```

Note, we use walker and pedestrian all over interchangeably to refer to the same thing.

Rong-Zhi, 2018-05-15:
When I set NumberOfPedestrians in CarlaSettings.ini to a larger number, let's say 100, the pedestrians spawned successfully, but most of them can't move at begining, it might take more than 5min for some of them starting to walk.  Can I change this behavior somewhere?

nsubiron, 2018-05-16:
@Rong-Zhi This seems to be an issue of Unreal's navigation system not being able to handle that many agents. It is a known issue that we don't know how to solve.

Rong-Zhi, 2018-05-16:
Ok... Thanks for your reply

YDSh510, 2019-05-15:
> "Walker Spawn Points" can only be edited from within Unreal Editor. You'd need to compile CARLA from source and open the CarlaUE4.uproject, then drag-and-drop the spawn points in the map.
>
>
> Controls [...]

@nsubiron - 
1. How can I add new  "Walker Spawn Points" in new Town 05 or 04. is it just by dragging "PlayerStart" or is there any other way.
2. Also If I have to use Player start , then how can I differentiate between vehicle start point or walker start point as both will have same Type.
3. Can we still control no. of pedestrian to spawn using CarlaSettings.ini  in Carla0.9.4?

nsubiron, 2019-05-15:
Hi @YDSh510,

We no longer use walker spawn points in 0.9.x versions, you can now spawn them anywhere in the map. At the moment they have to be controlled individually (direction + speed); we are working on a new navigation for pedestrians on the client-side.

As an example, you can control a pedestrian manually with our manual_control.py script

    ./manual_control.py --filter=walker

YDSh510, 2019-05-16:
@nsubiron  - Thanks for the info