carla-simulator/carla #92: How can I

18605973470, 2017-12-17:


felipecode, 2017-12-18:
Hello @18605973470 ,
Unfortunately that is a bit tricky to do now in CARLA, since the starting positions
have to be selected from a list ( Please see #14). 

However, by fixing the seed, the start positions of NPC are always the same.
 An alternative solution could be:

- Set a seed for npc vechicles  ( or leave the default), the npc car seeds can
be set on CarlaSettings parameter "SeedVehicles=123456789"

- Record the start position of the all the vehicles for this seed.

- Check the start position locations and select one that is close to the npc you want.

Eventually you could keep changing the npc vehicle spawn seeds to get a situation
where the start positions and the npc positions match your needs.

Cheers

18605973470, 2017-12-19:
Thanks a lot !

And I also want to know what's the method to generate index according to seeds given or where is the source code about that. Getting this, I can automatically find the NPC cars' start location instead of manually doing that.

nsubiron, 2017-12-19:
The spawning of non-player vehicles is done by "AVehicleSpawnerBase", there a random spawn pose is retrieved from the list of player starts present in the map.

18605973470, 2017-12-20:
In **VehicleSpawnerBase.cpp**, there is a function called **TryToSpawnRandomVehicle**. 
And I guess random index is generated here.
Digging into it, I find random number is generated through **std::uniform_int_distribution<int32>(Minimum, Maximum)(Engine)**.
Here Minimum is 0 and Maximum is 151. (start point list' length is 152)
And the **Engine** is defined as **std::minstd_rand Engine** in **RandomEngine.h** with seed 123456789 as default.
And the final random number is 145.

So I write a simple test code :
`std::minstd_rand Engine;
Engine.seed(123456789);
std::cout << std::uniform_int_distribution<int32_t>(0, 151)(Engine) << std::endl;`
But the result I get is always 8, instead of 145.

So I am confused .
What's wrong with me ?

nsubiron, 2017-12-20:
The implementation of the std random number engines is not defined by the standard and might vary from library to library. In fact, running that code from inside Unreal Engine produces 145 as first value
```
std::minstd_rand Engine;
Engine.seed(123456789);
UE_LOG(LogCarla, Error, TEXT("First random number is %d"), std::uniform_int_distribution<int32>(0, 151)(Engine));
```

Unreal is compiled using libc++ (instead of libstdc++ that Linux usually uses), you might try linking against libc++ in your external program and it might produce the same results.

closed 2018-01-09 by nsubiron